wavelog/wavelog #78: Please add support for PostgreSQL

beedaddy, 2024-02-02:
Since Postgresql is a very popular database system nowadays, I would like to ask for postgres support in the future. It would be really good to have a joice here, since running mysql just for one single service feels uncomfortable. I would also be happy to help with testing.

int2001, 2024-02-02:
Uh. that's a really huge Feature-Request.
All libraries i've seen for "Multi-DB-Support" are not well written. There are always performance-issues because you've to workaround. Perhaps that changed during the last years, but i don't think so.

And a complete switch to Postgres won't be possible, because many hams are using Hosters which don't provide postgres.

you're right in one point: PGSQL is the "better Database", but it is even more complex to configure for the User.

beedaddy, 2024-02-02:
Yes, I know it's would be a huge effort. Perhaps it could be considered again if there would ever be a major rewrite or so. At least many web application I know of do support both mysql and postgresql (Wallabag, Nextcloud, Roundcube, …). Having a choice would be really cool. As I said, maybe it's something worth to consider for the long term.

int2001, 2024-02-02:
yes, maybe an "enterprise Version" one day. But currently not on the Roadmap.
I know Roundcube/Wallabag an so on. Difference is: less SQL-Accesses and - tbh - not very performant there. Yes, depends on machine.