wavelog/wavelog #92: Undefined array key causes issues with installer

jk13xyz, 2024-02-04:
**Describe the bug**
I'm trying to dockerize Wavelog. When testing, the install script runs into an issue when starting the install. After about a second, it errors out with the following error:

> Warning: Undefined array key "database_check" in /var/www/html/install/index.php on line 94
> 
> Warning: Cannot modify header information - headers already sent by (output started at /var/www/html/install/index.php:94) in /var/www/html/install/index.php on line 130

The install script doesn't fail per se. When loading the base URL (which I think the header modification was trying), the site is busy loading without displaying anything. In the background, I can see the MySQL database being populated while the loading is taking place. The page eventually stops loading altogether. Upon another refresh, the login screen is displayed.

I reproduced this behaviour with different LAMP stack configs, as well as an nginx installation.

**Version Information**
- OS: Debian 12
- Apache: 2.4.57
- PHP: 8.2.15
- MySQL/MariaDB: 8.3
- Wavelog: 1.0 

**Steps to Reproduce**
1. Open installer,
2. Fill out all required fields
3. Hit install

HB9HIL, 2024-02-04:
Hello @jk13xyz 

Could you please try if you can reproduce this behaviour with MariaDB instead of MySQL?

jk13xyz, 2024-02-04:
Hi @HB9HIL 

The same behaviour occurs with MariaDB 11.2.2.

HB9HIL, 2024-02-04:
Can you explain exactly what you are trying to do? And how?

I can't reproduce this behaviour so it seems like you are causing a faulty "state" when running the installer. 

> Warning: Undefined array key "database_check" in /var/www/html/install/index.php on line 94

I have no explaination for this warning since the array key "database_check" should come as POST data when the install button is pressed. 

> Warning: Cannot modify header information - headers already sent by (output started at /var/www/html/install/index.php:94) in /var/www/html/install/index.php on line 130

This issue could be linked to the other one as the script tries to run again and the headers already set by the previous run. 

---

However atm I don't see a bug here because you try to dockerize something which does not offer docker support. If I try the installer in various combinations on a "normal" VM or bare metal, I can't reproduce this issue.

jk13xyz, 2024-02-05:
> Can you explain exactly what you are trying to do? And how?

I prefer using Docker for web apps and the like. Since there's none of Wavelog (yet), I figured I'd make one myself.

> I have no explaination for this warning since the array key "database_check" should come as POST data when the install button is pressed.

What kind of data should I expect when I look at the POST data? What I'm seeing are essentially all fields that I've entered (or the non-required ones I left blank as ""). Starting with "directory" and ending with "timezone". If "database_check" is indeed a POST value, it's not being transferred for whatever reason, unlike the rest.

> I can't reproduce this issue.

If you'd like, I can reset the instance on my server and let you walk through the installer. Contact me via Discord. Same username as my Github name.

HB9HIL, 2024-02-05:
Hi @jk13xyz 

I don't think thats necessary since docker support is not planned and I doubt it will be part of our roadmap. 

You can go through the code if the installer. The "database_check" value is transfered in line 692 which is part of the ajax call which triggers the database connection test in step number 4.

I'm converting this issue to a discussion as it is no bug in the installer. The installer seems to work properly in a non-docker environment.