wavelog/wavelog #105: Inconsistency in handling of station locations

205er, 2024-02-06:
Hi all,

I use station locations to separate my ham radio activities at different location. I have a station location for my home QTH, one for the club station and several locations for outdoor activities.
[image]

I always activate the station location, whenever I change my physical location to benefit from the prefilled values in QSO logging tab "station".
Unfortunately, the activation of station locations does not have an impact on the QSO overview (_Logbook_ --> _Overview_)
The QSO that are listed in this overview do NOT belong to the active station location, but to location with ID 1. This is misleading (and in my eyes also wrong), because the active station is indicated at the top left, but it does not relate to the below shown information (QSO).
[image]

I would propose to change this behavior: Always output the QSO in the overview that are related to the active station.

 Thanks for your amazing work
 br
 Falko

phl0, 2024-02-06:
Well it is showing the QSOs belonging to the logbook not the location. So it displays all QSOs of all locations that are linked to the active station logbook.

phl0, 2024-02-06:
I just did some tests and cannot reproduce the error. Did you link the OE location to the station logbook named "Standard Log"?

phl0, 2024-02-06:
See my test case here. I created a station profile named "test", linked it to my active station logbook called "All" and logged a QSO with DJ7NT:

[image]

The advanced logbook shows the QSO of the active station location. And in Logbook overview I can see the QSO:

[image]

Because the "test" location is linked to station logbook "All". Now if I remove the link the QSO disappears from the active logbook:

[image]

So I still think that your station location "OE" is not linked to station logbook "Standard Log".

phl0, 2024-02-06:
Added a change PR to remove station location from the listing as it is not what it is filtered for.

205er, 2024-02-06:
OK, OK, you are right! I finally got it! I did not link the location to my logbook, that's why the behavior of Wavelog was so strange.

closed 2024-02-07 by 205er