wavelog/wavelog #128: QSL Label printing

dg9vh, 2024-02-13:
**What Feature is Missing?**
When printing labels for modes that are submodes to a mode on the qsl label only the mode would be printed. For example: When using FT4 as submode the mode would be MFSK (what is correct) but the info that it was FT4 is lost.

**Why is This Feature Important?**
It's much more clearer what mode was used exactly for QSLing.

**Is This Feature Personal or Beneficial to Others as Well?**
I think this benefits all users that are logging QSOs via ADIF-import from digimode-programms for example.

int2001, 2024-02-14:
will check. tnx for input

int2001, 2024-02-14:
Check #134

phl0, 2024-02-14:
Checked and tested with 100% result :)

dg9vh, 2024-02-14:
Will confirm later this day after testing

dg9vh, 2024-02-14:
Confirmed as working :-) tnx!

closed 2024-02-14 by dg9vh