wavelog/wavelog #129: QSO-date and -time differs from lotw-last-upload

dg9vh, 2024-02-13:
**Describe the bug**

QSO-date and -time differ unplausibly from lotw-last-upload-timestamp of callsign, see screenshot:

[image]

Maybe an timezone-issue of the timestamp? Do we need here to convert the used timezone on the ARRL-system to UTC?

dg9vh, 2024-02-13:
To be a bit clearer: Here in this case, the QSO-partner confirmed a QSO visually on a time that was before the QSO happend :-)

dg9vh, 2024-02-13:
It is not a thing of conversion - I also have entries that differs more than 3 days into the past from the QSO-date... It looks like, that the last-update-timestamp from LOTW is not correctly updated when updating LOTW-data (via cron or manually). It seems that the state is saved at the moment, the QSO is logged and this data for this callsign is fetched for the first time.

HB9HIL, 2024-02-13:
The "Last Upload" data provided by LotW is not what we would call "Live". 

It's a CSV which is generated every few day's and served by LotW. So it can happen, that the User uploaded his Log, you got the confirmation but the `Last Upload` data is not updated. 

Unfortunately we can't solve that in Wavelog.

[https://lotw.arrl.org/lotw-help/developer-query-lotw-users/](https://lotw.arrl.org/lotw-help/developer-query-lotw-users/)

[image]

closed 2024-02-13 by HB9HIL