wavelog/wavelog #225: Storage engine MyISAM is disabled (Table creation is disallowed)

welly, 2024-03-21:
Hello,

I'm trying to get wavelog installed and have gone through the initial install process which eventually worked, however viewing the installed site reports this error:

```
An uncaught Exception was encountered

Type: mysqli_sql_exception

Message: Storage engine MyISAM is disabled (Table creation is disallowed).

Filename: /home/sites/htdocs/wavelog/system/database/drivers/mysqli/mysqli_driver.php

Line Number: 301
```

Wondered anyone has any suggestions on where to start to fix this? I'm using MySQL 8 on a Digital Ocean managed database.

Thanks

welly, 2024-03-21:
I created a fork and used InnoDB in the migrations, however I'm getting a new error which I'll create a new support ticket for.

closed 2024-03-21 by welly

int2001, 2024-03-22:
gm welly,
we're currently migrating most of the tables to innoDB. Fact is: 
The tables are initially created with myISAM and are later migrated to innoDB (one of the last migrations).

As far as i've done research digitalOcean does **not** support myISAM. (See https://www.digitalocean.com/community/questions/how-to-enable-myisam-for-mysql-managed-database ).
Therefore we cannot provide any "out of the box"-solution currently for this particular problem.

But: Perhaps we're going to refactor the installation completly to innodb (skip the way over myisam) in the future.

If you're familar with SQL/Databases, you can try the following as Workaround:
* Install Wavelog locallay with a myISAM-Enabled Version of MySQL (completly)
* Alter the leftover Tables to engine InnoDB
* Export the whole dump
* Import it at digitalOcean
* Don't forget to adjust the config-files, when moving to digitalOcean

HTH for the moment.
73 DJ7NT

welly, 2024-03-22:
Thanks very much @int2001, I'm going to give that a go. Big help, cheers!