wavelog/wavelog #227: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near '"bootstrap.min.css"' at line 2

welly, 2024-03-21:
Hello,

I've got a bit further with the install but am now getting this error. Not really sure where to start with this.

```
An uncaught Exception was encountered

Type: mysqli_sql_exception

Message: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near '"bootstrap.min.css"' at line 2

Filename: /home/sites/htdocs/wavelog/system/database/drivers/mysqli/mysqli_driver.php
```

The error is possibly referring to this:

https://github.com/wavelog/wavelog/blob/master/application/migrations/052_add_user_stylesheet.php#L10

Would welcome any guidance!

Thanks

int2001, 2024-03-22:
pse see my answer at #225 
perhaps the other way round works. (Creating everything at a myisam-enabled database, alter it, export/import)

int2001, 2024-03-22:
However. this seems to be an easy job. i am working on it at #228

welly, 2024-03-22:
Thanks very much! I'll try what you've suggested in the meantime until this gets merged but very much appreciated!

int2001, 2024-03-23:
@welly its now in the dev-branch. i'd be happy if you can confirm a "non-isam"-working Wavelog now...

closed 2024-04-03 by AndreasK79