wavelog/wavelog #269: Feature Request - Mail Address Management

wafer-li, 2024-04-12:
**By submitting feature requests, you have the opportunity to contribute your ideas and inspiration to the Wavelog Dev-Team. Please be aware that not every feature request will be addressed or answered.**

**What Feature is Missing?**

A Mail Address Management of each callsign

1. Dedicated page for Malling Address management of each callsign
2. Mailing Address Create, Edit, Delete
3. Search Address by callsign and text
4. Link callsign addresses to other callsign's, since the callsign could be different even if it is the same station. Such as `XXX` and `XXX/QRP`
5. Export the callsign and address as csv file with columns of "CALL, ADDRESS, ZIP_CODE"
6. Mailing Address update time and icon button to check callsign's QRZ.com and HAMQTH.com in case the address is expired.

**Why is This Feature Important?**

Checking someone's mailing address on QRZ.com or any other service will be tedious and require a lot of manual work.

Especially when you need to send a lot of QSL cards.

If the mailing address is stored and managed, it will relief the workload of checking the mailing address repeatedly.

**Is This Feature Personal or Beneficial to Others as Well?**

Collecting and Send/Receive QSL Card is a major part of Amateur Radio, even if there are so many third-party QSL system(like LoTW).

But how to make and write the QSL Card is always Lesson 101 of being a HAM.

I believe that there is a large group of people which still stick to sending and receiving the physical QSL Card.

Finding mailing address of a station is a tedious job, if you need to send out 50 or more QSL cards.

HB9HIL, 2024-04-23:
Dear @wafer-li 

this is a difficult topic. I definitly see the use case for this, as I have exactly the same workflow as you. 

BUT we can not just save personal data like an address in the database of each user. This is probably against data and privacy law. 

I'm sorry but we can't implement such an address storage.

closed 2024-04-23 by HB9HIL

wafer-li, 2024-04-24:
> Dear @wafer-li
>
> this is a difficult topic. I definitly see the use case for this, as I have exactly the same workflow as you.
>
> BUT we can not just save personal data like an address in the database [...]

OK, I will find another way to achieve it.

Thanks for your reply anyway.