wavelog/wavelog #270: Feature Request - offline mode of Contest Logging

wafer-li, 2024-04-12:
**By submitting feature requests, you have the opportunity to contribute your ideas and inspiration to the Wavelog Dev-Team. Please be aware that not every feature request will be addressed or answered.**

**What Feature is Missing?**
Offline mode of Contest Logging, just like the SFLE.

Some people like me don't have that much money to purchase a station room, or for a better signal and lower noise.

They must set up their station in the outside field, which lacks Internet connection.

In fact, I did my 2024 CQ WPX SSB contest in the outside field, with QRP, **pen and paper!**

It is painful and result of at least five duplicated QSOs, lost a ton of scores.

**Why is This Feature Important?**
Not every feature holds equal importance. Clarify why you think your suggested feature is crucial and how it enhances Wavelog.

**Is This Feature Personal or Beneficial to Others as Well?**
If the feature is solely for your benefit, and not beneficial to the broader Wavelog community, we may not prioritize or respond to your request. Ensure that your proposed feature contributes to the entire Wavelog user base.

HB9HIL, 2024-04-12:
Dear @wafer-li 

Offline mode was discussed in during our previous work for Cloudlog before we forked to Wavelog. We had some people which asked for an "Offline Mode" regarding a Progressive Web App in mobile devices. 

Unfortunately this is nearly impossible for us to achive such an offline feature since it would require sooo much more background services and programming an application which is absolutely not in an affordable ration between work and use-cases. 

There are some nice offline logging programms available you can use out in the field or sometimes its even easier to log on paper and tranfer the QSO with SimpleFLE.

I'm sorry that I can't give you an better answer. But with our current architecture (PHP, MySQL) it's not doable for us. 

I'm really sorry

closed 2024-04-12 by HB9HIL

HB9HIL, 2024-04-12:
PS: I will check if I can implement some contest features in SimpleFLE for Serial Number and Exchange.

wafer-li, 2024-04-13:
Sorry for the late reply, I went outside to CQ POTA this morning.

The contest feature, what I do really need is "duplicated QSO checking", that's why I propose to add the offline mode to contest logging.

Or maybe I should really spend money to purchase a laptop with long battery life.

Nevertheless, thanks for the reply, I will figure out the way for outside field contesting in some days.