wavelog/wavelog #271: Bug Repot - Advanced Search result in 500 internal error

wafer-li, 2024-04-13:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
/search/run_query return 500 internal error, but other search approach results good.

It does not seem to a DB issue, much like the php logic code error.

Because when run the query, it does provide at least one result in the response, but the HTTP Code is 500 and no result output to the page.

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:       Ubuntu22.4     
- PHP:      8.1       
- MySQL/MariaDB:    5.5.5-10.10.6-MariaDB-1:10.10.6+maria~ubu2204
- Wavelog:      1.4

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'https://<your.domain.com>/search/filter'
2. Click on 'Run query or Search' with any rules that will provide non-empty result
4. See error

**Expected behaviour**
It should show the result correctly.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser Chrome 
 - Version 123.0.6312.107

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Similar bug with cloudlog before I migrate to wavelog https://github.com/magicbug/Cloudlog/issues/3070

wafer-li, 2024-04-13:
Sorry, already fixed in `dev` branch.

Looking forward to see the next stable version.

closed 2024-04-13 by wafer-li

AndreasK79, 2024-04-13:
Yeah, we found the bug earlier😊 We will soon release a new version.