wavelog/wavelog #290: Language selection for the map

dl2de, 2024-04-16:
**By submitting feature requests, you have the opportunity to contribute your ideas and inspiration to the Wavelog Dev-Team. Please be aware that not every feature request will be addressed or answered.**

**What Feature is Missing?**
A language selection for the map or the set language of wavelog should also affect the map.

**Why is This Feature Important?**
If all the countries and places on the map are written in the respective national languages, most users will not be able to understand much, especially with different characters (such as Chinese, Cyrillic alphabet, Arabic script).

**Is This Feature Personal or Beneficial to Others as Well?**
Very few users will understand all the alphabets in the world. It should therefore be a very useful feature for most people to have the map in their own language or at least in English.

int2001, 2024-04-17:
There's currently no Maptile-Provider which provides labels in different languages.
If anybody has an idea about a "Internation-Tile-Server" which is free, we can implement it.

For the German-Users, there's an alternative. Simply connect to your MySQL and fire this SQL:

`UPDATE options SET option_value = 'https://tile.openstreetmap.de/tiles/osmde/{z}/{x}/{y}.png' WHERE option_name='map_tile_server';`

After that you're on a german map with bilingual labels (Native and german).

CAUTION: We're working on it, that the provider is also taken at logbook-advanced. the current SQL has no impact on the map at logbook-advanced but to most other maps.

HB9HIL, 2024-04-23:
@dl2de can we close this?

dl2de, 2024-04-24:
Yes

dl2de, 2024-04-24:
Yes

closed 2024-04-24 by dl2de