wavelog/wavelog #291: Image erstellter, erstellt ungewollten Rand im Bild

TheMockTv, 2024-04-17:
- OS:   Debian 12.2
- PHP:  8.2.7
- MySQL/MariaDB:    10.11.4-MariaDB-1~deb12u1
- Wavelog: 1.5.2

**Steps to Reproduce**
1. logbuch
2. Click on 'Erweitert'
3.Click on 'Karte'
4.Click on 'print map Current Size'
5. Save the Image 
6. Open Image
7. See error

**Expected Behavior**
Ein Bild in voller Größe ohne Rand

**Screenshots**

so bekomme ich das bild 
[image]

So sollte es sein
[image]

**Desktop Information:**
- OS: [Windows 10 Pro]
- Browser: [Opera]
- Version:  [108.0.5067.29] 
- Chromium-Version: [122.0.6261.112]

int2001, 2024-04-17:
Should be solved now with #292

HB9HIL, 2024-05-23:
@TheMockTv We just release Version 1.6. Please test and close this issue if you are happy with it :)

PR #292 (2024-04-17): Fix for Fullscreen-Export
This one fixes the Fullscreen-Export at Maps.

Old: Ugly bars at right and bottom side
New: Bars gone

Should resolve #291