wavelog/wavelog #320: Unable to go pass installation

N3evin, 2024-04-29:
Hi team, I did a clean installation and everything from scratch. However I keep getting 
`Not all fields have been filled in correctly. The host, username, password, and database name are required.`

The database info are working as the "connection test" passed. But when i click install, it keep throwing that error.

N3evin, 2024-04-29:
> hello, the email field is necessary too. i started a fix few days ago ... i have to finish it ;)

thank you, but I did add my email.

N3evin, 2024-04-29:
I think i might had it resolved, but callsign or locator seems to be required. Or else it throw the error.

HB9HIL, 2024-04-29:
@N3evin so it's resolved?

N3evin, 2024-04-29:
> @N3evin so it's resolved?

Yes it had been resolved, solution should be that it required a callsign

closed 2024-04-29 by N3evin