wavelog/wavelog #323: QSL message is not added when importing adif file

gotoradio, 2024-04-30:
When importing an adif file, the default QSL Message is not added.
It is not shown in the advanced tab and neither in the exported adif.

Wavelog version 1.5.1
Browser: iPadOS, Safari

phl0, 2024-04-30:
Can you please show an example of the ADIF file you imported?

HB9HIL, 2024-04-30:
@phl0 I tested aswell. 

Reproduce:

- Set a eQSL Default Message in a Station Location
- Import an ADIF File with QSO's
- See in LBA that there is no eQSL Default Msg for these QSO

We have to discuss wherever the eQSL Default Msg is added to these QSO if one is defined in the station location or not. Imho it would not be wrong to add it. 

Maybe we could also ask @abarrau for his opinion as he built/introduced the eQSL default msg stuff.

HB9HIL, 2024-04-30:
So at the moment it's working as designed. But it would be also logical if the eQSL Default Msg is added to the imported QSO's if there is one defined in the Station Location

phl0, 2024-04-30:
There is more to this topic than visible at first glance. IMHO the ADIF standard does not correctly state whether the field is for incoming our outgoing eQSL message. See my post here https://groups.io/g/adifdev/topic/104884450#msg9357 which has not been fully answered till now ...

HB9HIL, 2024-05-03:
Well we have an deeper issue here. In my honest opinion we should not wait with resolving this issue here just with waiting for a new ADIF spec. 

One possible solution and my recommendation would be:

Let's fill the QSLMSG field in ADIF import with the eQSL DefaultMsg from the station location, if this field is empty in the ADIF. This is what I would expect as a user. 

I think this is the best solution in this case because we have the fence that a user can only upload ADIFs which have a matching station callsign. So it theoretically can't be another operator with another QSLMsg. If there is already a QSLMSG in the ADIF then because the user already set it in another Log Software.

abarrau, 2024-05-04:
hello all, 

(sorry for delay of answer)
yes in fact, when I implemented the eQSL DefaultMsg functionality, the objective was for the final field to be filled automatically each time a QSO is created (if the eQSL DefaultMsg is defined).
So when adif is imported, if QSLMSG is empty, eQSL DefaultMsg is used for create QSO; it was forgotten on my part.

I can watch it as soon as I have some time.

73