wavelog/wavelog #345: Integration of DARC QSL Shop

Loki1737, 2024-05-07:
**What Feature is Missing?**
If I select QSL Cards to be printed, it would be a nice feature, if they could be automatically send to the DARC QSL (Printing) Service.

**Why is This Feature Important?**
It reduces, the hurdles, when you want to print QSL Cards

**Is This Feature Personal or Beneficial to Others as Well?**
I think the user base might have a difference in the need. It might not be the highest priority, but a nice to have feature.

int2001, 2024-05-07:
Currently you're able to export a ADIF for only requested (or you marked as requested) QSOs at "Labels -> Pending QSLs -> View QSOs" at your PC.
After that export you can upload that ADIF to the DARC-QSL-Service, and mark them all as sent within Wavelog.

That is one possible Workflow for using the QSL Service.

As far as i know the DARC-Service offers NO explicit interface for an automatic upload. But if you have any information about such an interface, feel free to share.

Loki1737, 2024-06-07:
The DARC Service Team responded to me, that they do not provide any API jet. But they are open for the idea. They would be interested to know, which features would be needed for the API. 

Is there a way, how I can support on this? Unfortunatly I do have limited coding experience, but system architecting experience. If it helps, I could generate a draft (diagram), what I think makes sense, but as I do not know the code, it might not build the whole picture. 

BR,
Lucas

Loki1737, 2024-06-07:
Okay, thank you very much @int2001 . Let me know, if I can help. And yes, I expected as well, that it will take its time.

BR,
DO5LE

HonkXL, 2026-10-26:
This would be a great feature. Because I had an other problem with uploading ADIF files there, I asked at QSL-Shop for this feature and got this reply:

"Das die WAVELOG vom DARC ist schon dabei eine Schnittstelle zum QSLSHOP zu programmieren. Wann diese fertig wird, kann ich leider nicht sagen."

Don't understand the reply, because they say wavelog is programming a API and Wavelog says the API is missing on QSL-Shop...

int2001, 2026-10-26:
lol.
sounds like circle-jerk. Without proper Interface from DARC, we cannot do anything here. as far as i know, it's still pending there.