wavelog/wavelog #377: Opening P12 unsupported while uploading LotW certs // Ubuntu + OpenSSL

Erklaeranlage, 2024-05-19:
**Description**
While uploading a .p12 LotW cert the upload form returns without error in UI. Nothing gets imported.
Cause is unsupported (legacy) routines in OpenSSL3.  Workaround is activating legacy modes, but that should not be permanent in my opinion.

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Ubuntu 22.04.4
- PHP:              8.1.2
- MySQL/MariaDB:    irrelevant for the bug
- Wavelog:          1.5.2
- OpenSSL 3.0.2

**Steps to Reproduce**
1. Have a default Ubuntu 22.04.4 installation
2. Upload certificate
3. see form return without error and without uploading the cert to the database

**Expected Behavior**
Have the .p12 certificate uploaded to the database.

**Resolution**
The underlying error is "error:0308010C:digital envelope routines::unsupported" which unfortunately does not get shown on the UI.
To resolv this issue, /etc/ssl/openssl.cnf has to be changed:
Replace the [default_sect] section of the file with:
`[default_sect]  
activate = 1  
[legacy_sect]  
activate = 1`  

and change the [provider_sect] section to:
`[provider_sect]  
default = default_sect  
legacy = legacy_sect`

Restart the machine and everything works.

73 de DB4SCW

phl0, 2024-05-19:
The legacy stuff is not needed anymore. Update your tqsl and export the certificate agn. It will then use up to date crypto which should upload in wavelog just fine.

Details here:

https://github.com/magicbug/Cloudlog/issues/2068

Erklaeranlage, 2024-05-20:
That's it. thank you!

closed 2024-05-20 by Erklaeranlage