wavelog/wavelog #393: Advanced logbook is very slow if it contains a lot of qsos

derFogel, 2024-05-23:
Hi,
first thanks for your awesome log software.
We are using wavelog in our clubstation. Recently we notices that the advanced logbook got really slow (5s with 25k qsos). Some 
profiling revealed the culprit. It is the following query from `application/models/Logbookadvanced_model.php`

```sql
SELECT *  FROM TABLE_HRD_CONTACTS_V01 qsos
INNER JOIN station_profile ON qsos.station_id=station_profile.station_id
LEFT OUTER JOIN satellite ON qsos.COL_SAT_NAME = satellite.name
LEFT OUTER JOIN dxcc_entities ON qsos.col_dxcc=dxcc_entities.adif
LEFT OUTER JOIN lotw_users ON qsos.col_call=lotw_users.callsign
LEFT OUTER JOIN (select count(*) as qslcount, qsoid from qsl_images group by qsoid) x on qsos.COL_PRIMARY_KEY = x.qsoid      
WHERE station_profile.user_id =  3 AND qsos.station_id = '2'
ORDER BY qsos.COL_TIME_ON desc, qsos.COL_PRIMARY_KEY desc
LIMIT 250;
```
Profiling the query with mariadb revealed the following.
```
+------------------------+----------+
| Status                 | Duration |
+------------------------+----------+
| Starting               | 0.000229 |
| checking permissions   | 0.000010 |
| Opening tables         | 0.000062 |
| After opening tables   | 0.000015 |
| System lock            | 0.000007 |
| table lock             | 0.000010 |
| Opening tables         | 0.000017 |
| After opening tables   | 0.000003 |
| System lock            | 0.000004 |
| table lock             | 0.000575 |
| Unlocking tables       | 0.000006 |
| closing tables         | 0.000017 |
| init                   | 0.000280 |
| Optimizing             | 0.000059 |
| Optimizing             | 0.000007 |
| Statistics             | 0.000041 |
| Preparing              | 0.000020 |
| Creating tmp table     | 0.000033 |
| Sorting result         | 0.000028 |
| Statistics             | 0.000330 |
| Preparing              | 0.000050 |
| Creating tmp table     | 0.000404 |
| Sorting result         | 0.000073 |
| Executing              | 0.000004 |
| Sending data           | 0.066014 |
| Executing              | 0.000005 |
| Sending data           | 0.000038 |
| Creating sort index    | 0.000051 |
| Removing tmp table     | 0.000006 |
| Creating sort index    | 4.602838 |
| Creating sort index    | 0.073452 |
| Removing tmp table     | 0.015294 |
| Creating sort index    | 0.000026 |
| End of update loop     | 0.000020 |
| Query end              | 0.000003 |
| Commit                 | 0.000006 |
| closing tables         | 0.000002 |
| Removing tmp table     | 0.000006 |
| closing tables         | 0.000002 |
| Unlocking tables       | 0.000001 |
| closing tables         | 0.000016 |
| Starting cleanup       | 0.000003 |
| Freeing items          | 0.000027 |
| Updating status        | 0.000031 |
| Reset for next command | 0.000006 |
+------------------------+----------+
```
The normal logbook and the callsign search are still working normally.

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Debian 12
- PHP:              8.2.18
- MySQL/MariaDB:    10.11.6-MariaDB-0+deb12u1
- Wavelog:          1.5.2

AndreasK79, 2024-05-23:
Hi,

Happy you like it, and thanks for the issue report. We did notice a slowdown, and a fix for that is in the dev branch. We will shortly make another release with these fixes. You can either try the dev branch, or wait till the release.

Andreas.

AndreasK79, 2024-05-23:
1.6 was just released. Give it a try and it should be quicker in this release.

AndreasK79, 2024-05-23:
It can't be the query that does this. What kind of hardware are you running?

derFogel, 2024-05-23:
I am running on an intel xenon X5680 with 16 Gigs of ram.
According to mariadbs slow query the following query is the problem:
```sql
SELECT *, dxcc_entities.name as dxccname, mydxcc.name AS station_country, exists(select 1 from qsl_images where qsoid = qsos.COL_PRIMARY_KEY) as qslcount, contest.name as contestname
                        FROM TABLE_HRD_CONTACTS_V01 qsos
                        INNER JOIN station_profile ON qsos.station_id=station_profile.station_id
                        LEFT OUTER JOIN satellite ON qsos.COL_SAT_NAME = satellite.name
                        LEFT OUTER JOIN dxcc_entities ON qsos.col_dxcc = dxcc_entities.adif
                        left outer join dxcc_entities mydxcc on qsos.col_my_dxcc = mydxcc.adif
                        LEFT OUTER JOIN lotw_users ON qsos.col_call = lotw_users.callsign
                        LEFT OUTER JOIN contest ON qsos.col_contest_id = contest.adifname
                        WHERE station_profile.user_id =  3
                        AND qsos.station_id = '2'

                        ORDER BY qsos.COL_TIME_ON desc, qsos.COL_PRIMARY_KEY desc
                        limit 250;
```

HB9HIL, 2024-05-23:
There must be an issue with your database installation. If I run the exact same query on my virtual machine (8gb ram) and set the LIMIT to 2000 I got:

[image]

derFogel, 2024-05-23:
Do you have all your qsos in one station location?

int2001, 2024-05-23:
@derFogel can you provide a db-dump to us?! 
Furthermore I can offer you 1:1 support tomorrow. No guarantee to solve it instantly but to a start a deeper investigation?!

derFogel, 2024-05-23:
yes I can share one. Do you have an email address I can send it to? Since it is a from a club station I don't want to share it publicly because one of the (ex-)members could have a problem with that.

The 1:1 support would be nice. I have time after 4pm.

int2001, 2024-05-24:
Since github offers no direct message, pse take the contact-information (email) from my qrz-page https://www.qrz.com/db/DJ7NT
to get in touch / deliver the dump

Once we're connected by Mail, we can schedule direct support.

int2001, 2024-05-24:
@derFogel seems like we found the issue. it's a missing index.
i fixed it at #401 - u can try it via two ways:
- switch to dev branch (when merged)
- if you are familar with SQL - Try this one:
```
CREATE INDEX `idx_contest_adifname`  ON `contest` (adifname) COMMENT '' ALGORITHM DEFAULT LOCK DEFAULT
```

AndreasK79, 2024-05-24:
@derFogel the reason I did not see the problem here, was that on my dev, I already had an index from when testing stuff earlier. Sorry about that, and thanks for reporting!

derFogel, 2024-05-24:
Thanks, after running the command it now takes only 1,3s (in firefox to load).

int2001, 2024-05-24:
yeah. so we can close the issue?!

closed 2024-05-24 by derFogel