wavelog/wavelog #402: Docker container updates but does not run latest migration.

anthonydiiorio, 2024-05-24:
Hi! I noticed a bug after pulling the latest Docker container. The migration file is not updated and needs to be manually edited to update Wavelog. 

Steps to reproduce:
1. Pull latest version
2. Open Wavelog

-  Version info still shows old version

[image]

-  New tools are shown in the menu like **Cron Manager** but result in `HTTP Error 500`

[image]

5. Manually edit migration.php
```
docker exec -it wavelog-main bash
apt install nano
nano /var/www/html/application/config/docker/migration.php
```

- Showing the difference between **config/migration.php** and **config/docker/migration.php**

[image]

6. Open Wavelog and it updates successfully.

[image]

**Environment:**
As described in the docker-compose.yaml in the wiki:
https://github.com/wavelog/wavelog/wiki/Installation-via-Docker

Thanks!

anthonydiiorio, 2024-05-24:
[image]

HB9HIL, 2024-05-24:
Hello @anthonydiiorio 

you usually don't have to edit the migration.php manually. We run test machines with docker aswell to test the release migration and here everything worked normally. 

You also don't need to run the cron manually in the docker version. Check the Cronmanager. The cron is already set up properly. ( The correct url would be `.../cron/run`, in your screenshot you used `.../cron`

substa, 2024-05-24:
I confirm the problem.
Fixed editing manually migration.php

HB9HIL, 2024-05-24:
please take a look aswell @int2001

HB9HIL, 2024-05-24:
@substa @anthonydiiorio 

WHEN did you both repull the image? We had a little issue with the tagging after the release yesterday.

anthonydiiorio, 2024-05-24:
Container updated automatically with Watchtower but I repulled manually about an hour ago just to be sure.

substa, 2024-05-24:
I updated by re-pulling the image from Portainer shortly after the release came out. I checked ghcr.io/wavelog/wavelog:latest, and it had already been updated about two hours earlier.

Additional information:

My migration_version was different from @anthonydiiorio 's. If I remember correctly, it was 192, but I'm not sure about this (I'm on stable branch). 
Besides the issues Antonio encountered (500 error on the cron page and debug, release version stuck at the previous one on the version info page), I wasn't able to receive data from QRZ after entering the callsign. 
After manually modifying the file, everything works without any issues.

int2001, 2024-05-24:
wait.
those `./application/config/docker/migration.php` - where does it come from?
it isn't shipped by us, because it has to live in `./application/config/migration.php`. and the docker-subfolder is a local volume (which isn't updated by a docker pull).

i have to dig in deeper later to that. but would be great to describe exactly what happened here to us:

- 1st installation with docker. When did you do that (because we changed the location of config.php during the 1st few feeks of experimental docker-support)
- adjustments you made to the docker-compose.yaml
- adjustments you made to config.php / migration.php / moving files

PS: 1.6. has Mig 202

HB9HIL, 2024-05-24:
@anthonydiiorio @substa 

additionally: 

in the folder `application/config/docker` or in the corresponding mapped volume should be only two files:

- config.php
- database.php

nothing else
[image]

substa, 2024-05-24:
> * 1st installation with docker. When did you do that (because we changed the location of config.php during the 1st few feeks of experimental docker-support)

Migration from cloudlog, march 2024.
Previosly using your docker image from here:
https://github.com/int2001/wavelog_docker
then updated to use the latest "official" docker-compose.yaml, from
https://github.com/wavelog/wavelog/wiki/Installation-via-Docker

Currently in application/config/docker I have many more files than those listed by @HB9HIL , probably due to the previous migration?

`
root@3fbe02891122:/var/www/html/application# ls config/
autoload.php  config.sample.php  database.sample.php  doctypes.php       hooks.php   lotw.php       mimes.php     routes.php   user_agents.php  wavelog_data_lists.php   bands.php     constants.php      docker               foreign_chars.php  index.html  migration.php  profiler.php  smileys.php  wavelog.php

root@3fbe02891122:/var/www/html/application# ls config/docker/
autoload.php  config.sample.php  database.sample.php  hooks.php   migration.php  routes.php       wavelog.php
bands.php     constants.php      doctypes.php         index.html  mimes.php      smileys.php      wavelog_data_lists.php
config.php    database.php       foreign_chars.php    lotw.php    profiler.php   user_agents.php
 `

int2001, 2024-05-24:
Ahh.
Roger.

That was the old Version (the docker-image from my personal repo).

And your docker-compose.yaml is exactly the same like in the wiki now?

The thing is: in the first version the complete tree at `./application/config` was a volume.
We changed that when migration to the main-repo happened that only `./application/config/docker` is the volume.

This is the reason for so many files at `./application/config/docker` on your system.

you have to make sure, that the config-volume is ONLY mounted to `./application/config/docker` in your container. Not to `./application/config`.
The `./application/config` holds files which are updated (like `migration.php`). If your volume is mounted to `./application/config` those migration.php will never be up2date.

A possible manual process could be:
- Save `config.php` and `database.php` locally.
- Stop the stack: `docker-compose down`
- Destroy the containers (`docker container rm wavelog-main && docker container rm wavelog-db` - or whatever your containers are names)
- Double-Check the docker-compose.yaml from the wiki
- Rebuild everything (normally the database-contents should survive this)
- Start it (`docker-compose up -d`)
- copy those two saved files back to `./application/config/docker/`

HTH