wavelog/wavelog #412: Enabled QSL-Methods not shown correctly in Logbook Overview

zone11, 2024-05-26:
**Describe the bug**
With having one or multiple QSL-Methods enabled in the account, disabling QSL results in hiding the LoTW column instead of QSL column in Logbook overview.
In this example, only enabling LoTW is described.

**Version Information**
- OS:               FreeBSD
- PHP:              8.1
- MySQL/MariaDB:    10.6.16
- Wavelog:          1.6 

**Steps to Reproduce**
1. Go to 'Account'
2. Click on 'Default Values'
3. Enable "LoTW" only
4. Got to Logbook -> Overview
5. See the QSL column instead of LoTW

**Expected Behavior**
Only show the enabled QSL-Methodes columns

**Screenshots**
Settings
[image]

Logbook Overview
[image]

**Desktop Information:**
- OS:               MacOS 14.5
- Browser:         Chrome / Safari
- Version:          17.5 / 125.0.6422.76

int2001, 2024-05-26:
Can confirm the bug.
Working on it. Tnx for reporting

zone11, 2024-06-14:
Tnx for fixing!

PR #414 (2024-05-27): Fixes QSL-Columns
Should Fix #412
closed 2024-06-14 by zone11