wavelog/wavelog #418: Add the ability to set a default radio used for live QSO

zone11, 2024-05-27:
**What Feature is Missing?**
I would recommend adding the ability to define a default radio (from the hardware interface list) to be used on the "Live QSO" page.

**Why is This Feature Important?**
Using hardware interfaces is an awesome feature, but every time you navigate around Wavelog, coming back to add a live QSO, you need to set the radio first. The feature would save time getting ready, especially in hectic moments.

**Is This Feature Personal or Beneficial to Others as Well?**
It's definitely a benefit for every user of a hardware interfaces in Wavelog

wafer-li, 2024-06-03:
And maybe adding the "default ANTENNA" as well?

HB9HIL, 2024-08-05:
@zone11 

Started to work on this. PR #700 is still drafted but feel free to test.

zone11, 2024-08-08:
@HB9HIL I've done a quick functional test, works fine and the implementation is very user friendly. Tnx!

HB9HIL, 2024-08-09:
The "default radio" handler is now available in `dev` branch and will be published in the next release.

zone11, 2024-08-09:
Thank you for the implementation!

PR #700 (2024-08-09): Default Radio
Ability to set a default radio from the hardware interfaces list. 

[image]
[image]

Behaviour:
- if no default radio is set:
   Switch to QSO logging / Contesting -> no radio selected by default
   Log a QSO -> last used radio get's written to session data
   Go to another page and come back to QSO logging / Contesting
   Last used radio is already selected
   Radio is lost on logout / re-login

- if a default radio is set:
   Switch to QSO Logging / Contesting -> default radio is already set
   log a QSO -> last used radio get's written to session data
   Go to another page and [... cut]
closed 2024-08-09 by zone11