wavelog/wavelog #432: printmap only white background

ds2k5, 2024-05-30:
Hello,
using Version 1.6

When using "printmap" got only a white background.

[image]

[image]

What did I wrong ?

HB9HIL, 2024-05-30:
Can you please try that in "Logbook Advanced" aswell?

[image]

Click "Map" and print the map from there.

ds2k5, 2024-05-30:
@HB9HIL   the same issue

[image]

HB9HIL, 2024-05-30:
Did you change the maptile server manually?

ds2k5, 2024-05-30:
>Did you change the maptile server manually?

no did not know how :-)

ds2k5, 2024-06-07:
Thank you.... disabled all in Firefox - no change

Tryed: chromium there I works

closed 2024-06-07 by ds2k5

HB9HIL, 2024-06-07:
> Thank you.... disabled all in Firefox - no change
> 
> Tryed: chromium there I works

Firefox is sometimes kinda tricky