wavelog/wavelog #433: no distance analyse will be show

ds2k5, 2024-05-30:
Hi,
using Version 1.6
have to QSOs added but no data in the distance analayse.
What did I wrong ?

[image]

Can see the 2 QSOs in ../wavelog/index.php/statistics

[image]

HB9HIL, 2024-05-30:
Hello @ds2k5 

You can check two things. 

- First make sure that the QSO does hava a valid grid locator and your station location does have a valid gridsquare aswell
- Go to "Admin -> Update Country Files -> Second Tab "Distance Data""

Here you can Wavelog let recalculate missing distance data. 

Try that.

ds2k5, 2024-05-30:
Thanks a lot ! sorry my fault! did not set the gridsquare!

closed 2024-05-30 by HB9HIL