wavelog/wavelog #443: [Cron Manager] All cron jobs fail

dl3jop, 2024-06-01:
**Describe the bug**
I enabled cron manager but all jobs are failing. The next time the job is supposed to run gets calculated but on the exact date only errors are shown in my log output of my cronmanger cronjob

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Debian 12, Kernel 6.1.85
- PHP:             8.1.27
- MySQL/MariaDB:    10.11.6
- Wavelog:         1.6 latest commit (d32293e in dev branch)

**Steps to Reproduce**
1. Enable cronmanger in crontab, disable old cronjobs
2. Setup cronjobs on cronmanager web interface as desired
3. Wait for a job to be triggered
4. Watch output file `wl_log` of  `curl --silent https://192.168.1.41/index.php/cron/run &>/root/wl_log`
Example output for eqsl_sync and clublog_sync:

`CRON: clublog_upload -> is due: true
CRON: clublog_upload -> RUNNING...
ERROR: Something went wrong with clublog_upload
CRON: eqsl_sync -> is due: true
CRON: eqsl_sync -> RUNNING...
ERROR: Something went wrong with eqsl_sync
`

Clublog as well as eqsl sync were working flawlessly using the old cronjobs stated in the crontab file.
Wavelog log threshold is set to 4 and the output truncated to the last minute:

`INFO - 2024-06-01 13:50:01 --> Config Class Initialized
INFO - 2024-06-01 13:50:01 --> Hooks Class Initialized
INFO - 2024-06-01 13:50:01 --> UTF-8 Support Enabled
INFO - 2024-06-01 13:50:01 --> Utf8 Class Initialized
INFO - 2024-06-01 13:50:01 --> URI Class Initialized
INFO - 2024-06-01 13:50:01 --> Router Class Initialized
INFO - 2024-06-01 13:50:01 --> Output Class Initialized
INFO - 2024-06-01 13:50:01 --> Security Class Initialized
INFO - 2024-06-01 13:50:01 --> Input Class Initialized
INFO - 2024-06-01 13:50:01 --> Language Class Initialized
INFO - 2024-06-01 13:50:01 --> Loader Class Initialized
INFO - 2024-06-01 13:50:01 --> Config file loaded: /var/www/html/application/config/wavelog.php
INFO - 2024-06-01 13:50:01 --> Config file loaded: /var/www/html/application/config/bands.php
INFO - 2024-06-01 13:50:01 --> Config file loaded: /var/www/html/application/config/lotw.php
INFO - 2024-06-01 13:50:01 --> Helper loaded: url_helper
INFO - 2024-06-01 13:50:01 --> Helper loaded: security_helper
INFO - 2024-06-01 13:50:01 --> Helper loaded: language_helper
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/account_lang.php
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/adif_lang.php
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/admin_lang.php
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/awards_lang.php
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/contesting_lang.php
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/eqsl_lang.php
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/export_lang.php
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/filter_lang.php
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/general_words_lang.php
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/gridsquares_lang.php
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/lotw_lang.php
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/menu_lang.php
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/notes_lang.php
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/options_lang.php
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/qslcard_lang.php
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/qso_lang.php
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/station_lang.php
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/statistics_lang.php
INFO - 2024-06-01 13:50:01 --> Database Driver Class Initialized
INFO - 2024-06-01 13:50:01 --> Session: Class initialized using 'files' driver.
INFO - 2024-06-01 13:50:01 --> Migrations Class Initialized
INFO - 2024-06-01 13:50:01 --> Language file loaded: language/english/migration_lang.php
INFO - 2024-06-01 13:50:01 --> Database Forge Class Initialized
INFO - 2024-06-01 13:50:01 --> Model "Options_model" initialized
INFO - 2024-06-01 13:50:01 --> Model "User_options_model" initialized
INFO - 2024-06-01 13:50:01 --> Model "Logbooks_model" initialized
INFO - 2024-06-01 13:50:01 --> Model "Stations" initialized
INFO - 2024-06-01 13:50:01 --> Model "Oqrs_model" initialized
INFO - 2024-06-01 13:50:01 --> Controller Class Initialized
INFO - 2024-06-01 13:50:01 --> Model "Cron_model" initialized
INFO - 2024-06-01 13:50:01 --> Final output sent to browser
INFO - 2024-06-01 13:50:01 --> Total execution time: 0.0410
`

HB9HIL, 2024-06-01:
@dl3jop 

Please check if all necessary `php` modules are installed. At the first glance it looks like a `php` module is missing. Please check in `Admin -> Debug` for missing modules.

dl3jop, 2024-06-01:
@HB9HIL Thanks for the quick answer, it seems like all modules are installed:
[image]

HB9HIL, 2024-06-01:
@dl3jop send me an email pls. mail on qrz

dl3jop, 2024-06-01:
Closed as per explanation by @HB9HIL:

https://github.com/wavelog/wavelog/blob/0b8e38be2e6e962ad4f140a4de4676941f01705d/application/controllers/Cron.php#L97

curl fails if the base url uses an untrusted cert (--insecure option can solve this). For now, switching to plain http mitigates the issue.

Tnx for the quick help!

closed 2024-06-01 by dl3jop

HB9HIL, 2024-06-03:
@dl3jop This issue will also be solved if #339  is finished. Then we can call the functions of cronmanager directly in PHP and get rid of curl.