wavelog/wavelog #446: Map in home page not plotting QSO correctly

wafer-li, 2024-06-03:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
Map in home page should plot the QSO use the current active station location as center.

Check the screenshot for details, the QSO with USA station not plotted at the right side of the map.

But the map in "Advanced logbook" is correct.

**Steps to Reproduce**
1. Go to 'demo.wavelog.org'
2. Login, make, link and active a station located in China.
3. Add a QSO with USA station
4. Go back home page
5. See error

**Expected Behavior**
The QSO at map should be plotted as the "Advanced logbook" map 

**Screenshots**
Include screenshots if applicable to illustrate the issue.

[image]

int2001, 2024-06-05:
Tried to reproduce it. Wasn't successful.
Maybe you forgot to link the 2nd station (you created at china) to the mainlog?

[image]

wafer-li, 2024-06-05:
> Tried to reproduce it. Wasn't successful. Maybe you forgot to link the 2nd station (you created at china) to the mainlog?
> 
> <img alt="image" width="907" src="https://private-user-images.githubusercontent.com/1410708/336682217-c75db3a0-ea7c-4ab6-ad0a-f705596ea2bc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc1NzE3MTYsIm5iZiI6MTcxNzU3MTQxNiwicGF0aCI6Ii8xNDEwNzA4LzMzNjY4MjIxNy1jNzVkYjNhMC1lYTdjLTRhYjYtYWQwYS1mNzA1NTk2ZWEyYmMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDVUMDcxMDE2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YTBmOGEyMjljM2Y2MWE5YjVjYmJjMmZjZjIzNThmODlmYzZhYjUyOGZjMmQ1NmE4MmU5NTllZGVjZDBjMGRmZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.dNRaPhnR3AJDWhhyOe4hx_PljS87HqLKb2hxzkb7KKs">

I Change the grid in demo to ON80

You could see the map in "Advanced Logbook"

[image]

And map in home page

[image]

int2001, 2024-06-05:
Ahh. it's ONLY about centering. understand.

Perhaps @AndreasK79 can help?!

ds2k5, 2024-06-07:
Thank you.... disabled all in Firefox - no change

Tryed: chromium there I works

HB9HIL, 2024-06-26:
@wafer-li 

This now should be fixed in `dev` branch. You can test. Please give some feedback. 

PS: We need some chinese translations :) if you want to contribute please check out [our translation platform](https://translate.wavelog.org/engage/wavelog/) :)

wafer-li, 2024-07-09:
> @wafer-li
> 
> This now should be fixed in `dev` branch. You can test. Please give some feedback.
> 
> PS: We need some chinese translations :) if you want to contribute please check out [our translation platform](https://translate.wavelog.org/engage/wavelog/) :)

Sorry for the late reply, I was busy with a big project for my company.

I deploy the `dev` branch, but the problem still exist, see the pics.

[image]

[image]

I believe it is the `dev` branch, since the migration version is `207`, which `master` branch is `204`

[image]

----------------------------

For translation, I still have not much time to do it, sorry.

Maybe I will join it when I have spare time, it is so busy these days.

HB9HIL, 2024-07-09:
> > @wafer-li
> > This now should be fixed in `dev` branch. You can test. Please give some feedback.
> > PS: We need some chinese translations :) if you want to contribute please check out [our translation platform](https://translate.wavelog.org/engage/wavelog/) :)
> 
> Sorry for the late reply, I was busy with a big project for my company.
> 
> I deploy the `dev` branch, but the problem still exist, see the pics.
> 
> <img alt="image" width="1511" src="https://private-user-images.githubusercontent.com/12459199/346838769-e82efa65-e59f-4b68-98ea-458457d2ce19.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA1MTA3NDksIm5iZiI6MTcyMDUxMDQ0OSwicGF0aCI6Ii8xMjQ1OTE5OS8zNDY4Mzg3NjktZTgyZWZhNjUtZTU5Zi00YjY4LTk4ZWEtNDU4NDU3ZDJjZTE5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzA5VDA3MzQwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUyZjc1MjMyYjc3Mzg2YzU0NWQzZTk1ZTNiODA5MDJlZDViNjY5MmVkOGNjNjAwZDU3MjE2ZWNhOGIxNDgzMmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.t4A0UtZ2QFRiwyVh0CxGCEAVjp5KWuqWMhhdK4di_OM"> <img alt="image" width="1566" src="https://private-user-images.githubusercontent.com/12459199/346839146-bcaf4401-3ad5-473a-9d99-28c0712edc09.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA1MTA3NDksIm5iZiI6MTcyMDUxMDQ0OSwicGF0aCI6Ii8xMjQ1OTE5OS8zNDY4MzkxNDYtYmNhZjQ0MDEtM2FkNS00NzNhLTlkOTktMjhjMDcxMmVkYzA5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzA5VDA3MzQwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTljMWZiNzNlY2M5MTZmYzk4NTAyMTQzNzZjYmMwMGEwZjQyMTdkY2Q3OTZjMGQ1MTQyMzMxMzZjMDNhNDZmOWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.bb8W-N8LY03KasnvD_V5-ySGLSrXw7fBTJsJwhgtZKg">
> I believe it is the `dev` branch, since the migration version is `207`, which `master` branch is `204`
> 
> <img alt="image" width="634" src="https://private-user-images.githubusercontent.com/12459199/346839451-6b68c062-65fa-403e-9fb7-dde23ef3f6b4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA1MTA3NDksIm5iZiI6MTcyMDUxMDQ0OSwicGF0aCI6Ii8xMjQ1OTE5OS8zNDY4Mzk0NTEtNmI2OGMwNjItNjVmYS00MDNlLTlmYjctZGRlMjNlZjNmNmI0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzA5VDA3MzQwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU5YjY3YWY3MGJiNDU0YTJjMzIzZmQyMGRmNGVlODBlNTU5NWI5MmFmNTE0MzYwMDU5NGRiNWEzZWQxNzJhMzYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qOsIx9_12smF75lj_7F-OR9iSZ2eLhC4hugm3w3WkIE">
> For translation, I still have not much time to do it, sorry.
> 
> Maybe I will join it when I have spare time, it is so busy these days.

Did you try to clear your browser cache?

wafer-li, 2024-07-09:
> Did you try to clear your browser cache?

Oh!

After force refresh, the plotting is correct now:

[image]

Thanks for notify me the cache things.

HB9HIL, 2024-07-09:
> > > > @wafer-li
> > > > This now should be fixed in `dev` branch. You can test. Please give some feedback.
> > > > PS: We need some chinese translations :) if you want to contribute please check out [our translation platform](https://translate.wavelog.org/engage/wavelog/) :)
> > > 
> > > 
> > > Sorry for the late reply, I was busy with a big project for my company.
> > > I deploy the `dev` branch, but the problem still exist, see the pics.
> > > <img alt="image" width="1511" src="https://private-user-images.githubusercontent.com/12459199/346838769-e82efa65-e59f-4b68-98ea-458457d2ce19.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA1MTA3NDksIm5iZiI6MTcyMDUxMDQ0OSwicGF0aCI6Ii8xMjQ1OTE5OS8zNDY4Mzg3NjktZTgyZWZhNjUtZTU5Zi00YjY4LTk4ZWEtNDU4NDU3ZDJjZTE5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzA5VDA3MzQwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUyZjc1MjMyYjc3Mzg2YzU0NWQzZTk1ZTNiODA5MDJlZDViNjY5MmVkOGNjNjAwZDU3MjE2ZWNhOGIxNDgzMmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.t4A0UtZ2QFRiwyVh0CxGCEAVjp5KWuqWMhhdK4di_OM"> <img alt="image" width="1566" src="https://private-user-images.githubusercontent.com/12459199/346839146-bcaf4401-3ad5-473a-9d99-28c0712edc09.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA1MTA3NDksIm5iZiI6MTcyMDUxMDQ0OSwicGF0aCI6Ii8xMjQ1OTE5OS8zNDY4MzkxNDYtYmNhZjQ0MDEtM2FkNS00NzNhLTlkOTktMjhjMDcxMmVkYzA5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzA5VDA3MzQwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTljMWZiNzNlY2M5MTZmYzk4NTAyMTQzNzZjYmMwMGEwZjQyMTdkY2Q3OTZjMGQ1MTQyMzMxMzZjMDNhNDZmOWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.bb8W-N8LY03KasnvD_V5-ySGLSrXw7fBTJsJwhgtZKg">
> > > I believe it is the `dev` branch, since the migration version is `207`, which `master` branch is `204`
> > > <img alt="image" width="634" src="https://private-user-images.githubusercontent.com/12459199/346839451-6b68c062-65fa-403e-9fb7-dde23ef3f6b4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA1MTA3NDksIm5iZiI6MTcyMDUxMDQ0OSwicGF0aCI6Ii8xMjQ1OTE5OS8zNDY4Mzk0NTEtNmI2OGMwNjItNjVmYS00MDNlLTlmYjctZGRlMjNlZjNmNmI0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzA5VDA3MzQwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU5YjY3YWY3MGJiNDU0YTJjMzIzZmQyMGRmNGVlODBlNTU5NWI5MmFmNTE0MzYwMDU5NGRiNWEzZWQxNzJhMzYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qOsIx9_12smF75lj_7F-OR9iSZ2eLhC4hugm3w3WkIE">
> > > For translation, I still have not much time to do it, sorry.
> > > Maybe I will join it when I have spare time, it is so busy these days.
> > 
> > 
> > Did you try to clear your browser cache?
> 
> Oh!
> 
> After force refresh, the plotting is correct now:
> 
> <img alt="image" width="1164" src="https://private-user-images.githubusercontent.com/12459199/346875841-bc63789e-b86e-4ee2-86c8-c1ec42e2aa74.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA1MTM3NDAsIm5iZiI6MTcyMDUxMzQ0MCwicGF0aCI6Ii8xMjQ1OTE5OS8zNDY4NzU4NDEtYmM2Mzc4OWUtYjg2ZS00ZWUyLTg2YzgtYzFlYzQyZTJhYTc0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzA5VDA4MjQwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY3YTRjMGMxNzBjNjhlMWJlZjQxODFhNjEzMzc0MGJjOTBlNTQxNWUzODA3ZTkzMzljNDdiOWYwMTNmZDNjMGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.H03II_FjXMZjqYpbi72vdVYHCLGWWIsBeCwbpp_F3S8">
> Thanks for notify me the cache things.

Great :)

closed 2024-07-09 by HB9HIL