wavelog/wavelog #449: QSL Card in PDF Format

HSGEV, 2024-06-05:
**What Feature is Missing?**
QSL Card upload in PDF file format

**Why is This Feature Important?**
Today I can only upload QSL-Cards as a picture.  This is very complicated, because you have to create an image for the front and an image for the back side.  I scan my QSL Cards with a document scanner or printer. These automatically create a PDF with the front and back side in one file. 

**Is This Feature Personal or Beneficial to Others as Well?**
The feature ist beneficial for others, because many other OMs an Yes use the PDF format to archive QSL-Cards

HB9HIL, 2024-06-05:
Hello @HSGEV 

I never saw a scanner which can't create `png` or `jpg` files from a scanned document. PDF files are handy for multipage documents and other purposes. For QSL cards it's way more ressource efficient to use simple image files.

I scan my QSL cards with a Brother DS-740D which is really handy, fast and creates great quality image files.

HSGEV, 2024-06-05:
Hello @HB9HIL,

my scanner can create image files too. But it isn’t ressource efficient. A QSL card is has 2 pages. A front and a back side. So a pdf file is the perfect format to archive a QSL card. 

By the way a PDF file is smaller than two image files.

closed 2024-10-05 by HB9HIL