wavelog/wavelog #452: Callsign Lookup - No informations if a suffix like /P or a prefix is added to the Call

zone11, 2024-06-06:
**Describe the bug**
When adding a QSO (Live or Post-QSO) with a callsign using a suffix like /P, no informations will be added to the QSO. The same applies to prefixes for different countries. The fallback to the callsign without is made, but only shown as information.

**Version Information**
- Wavelog:          1.6 

**Steps to Reproduce**
1. Go to QSO -> Live QSO
2. Click in the callsign field and add CALL/P (e.g. HB9HJQ/P and LA/HB9HJQ)
3. Check if the name is added.
4. Check the informations in the box on the right below the map. 

**Expected Behavior**
If no direct match for CALL/SUFFIX or PREFIX/CALL is found, use the informations for CALL

**Desktop Information:**
- OS:               MacOS
- Browser:          Safari
- Version:          17.5

**Screenshot**
[image]
[image]

phl0, 2024-06-06:
Difficult question. IMHO except for the name none of the normally fetched information is usable if you enter a prefixed or suffixed call (i.e. DXCC suffix). So I guess it would only make sense to pull the name from the logbook?

zone11, 2024-06-06:
Good point, location data is not valid in both cases and could be ignored. The name would definitely make sense.

DK9JC, 2024-06-19:
i would also very much welcome this. the name based on the lookup at qrz etc. and as a backup the name you already have in the log with "homecall". QSL route - as already written - would also be good.

m0urs, 2024-06-29:
Das hätte ich auch sehr gerne. Das Problem hatten wir bei Cloudlog schon mal diskutiert. Und dort wurde es so implementiert:

Wenn bei QRZ.com ein entsprechender Eintrag existiert, dann werden die dort hinterlegte Daten aus QRZ.com übernommen.
Wenn nicht, wird versucht, das Original Callsign zu ermitteln, also statt "ON/DG3NAB" dann "DG3NAB" bzw. statt "DG3NAB/P" dann "DG3NAB", bzw. bei "ON/DG3NAB/P" zuerst "ON/DG3NAB" und dann auch "DG3NAB".

Das wäre wirklich sehr hilfreich, wenn das auch bei Wavelog wieder so funktionieren würde.

HB9HIL, 2024-07-06:
@zone11 @DK9JC @m0urs 

This issue should now be fixed in `dev` branch. Feel free to test and feedback.

@zone11 Please close the issue if it's solved for you :)

m0urs, 2024-07-06:
Thank you very much, for the quick fix. It does work well for me!

zone11, 2024-07-06:
Tested with every thinkable variation, works perfekt! Thank you very much!

closed 2024-07-06 by zone11