wavelog/wavelog #459: Add "Update from Callbook"-Botton in QSO-Details and Edit-QSO

dg9vh, 2024-06-09:
**What Feature is Missing?**
Please add an "Update from Callbook"-Button in QSO-Details and Edit QSO:
[image]

[image]

**Why is This Feature Important?**
Because many of us are using third-party-software (for example N1MM for contesting or WSJT-X for FTx-QSOs), we don't have detailed data on name, QTH and so on and with this button (that is already available in advanced logbook) one can fetch missing data from callbook-services.

This is especially of interrest, if you are viewing the QSOs in the QSL-print-list or if you search for single callsigns via search function to have this function on single-QSO-base to not have the necessarity to switch over to advanced log to do this.

**Is This Feature Personal or Beneficial to Others as Well?**
I think this feature would be a benefit for all users.

phl0, 2024-06-09:
Which fields should be filled? Same as LBA? What about already existing, i.e. logged fields?

dg9vh, 2024-06-09:
I think it would be a good start to have it exactly like the function in advanced logbook. Using the same backend-function but only for the one specific qso.

HB9HIL, 2024-07-01:
@dg9vh 

Hello Kim

I started with a draft for the button. Imho we don't need this button in two places ("QSO Details" AND "QSO Edit"). I implemented a basic button in QSO edit which fills all fields which are available from callbook response. Also it shows green borders of this fields (Todo here: only add border to changed fields)

Currently it overwrites the data. Maybe it should only fill fields which are empty. Need to change that. 

What do you think about this draft?

HB9HIL, 2024-07-09:
@dg9vh 

Your button is merged to `dev` branch. Will be in the next release. Feel free to close this Feature Request if this button works as expected :)

phl0, 2024-07-09:
Works (also with hamqth).

HB9HIL, 2024-07-21:
@dg9vh 

Can we close this? :)

closed 2024-07-24 by HB9HIL

HochdruckHummer, 2024-11-08:
Yes, this function is tested by me in 1.8.8. sucessfully.