wavelog/wavelog #466: Is the docker compose file tested?

izeroo, 2024-06-11:
I don't see how wavelog-main can access wavelog-db, in the installation stage it keeps saying "connection refused".
```
version: '3'
services:
  wavelog-db:
    image: mariadb:11.3
    container_name: wavelog-db
    environment:
      MARIADB_RANDOM_ROOT_PASSWORD: yes
      MARIADB_DATABASE: wavelog
      MARIADB_USER: wavelog
      MARIADB_PASSWORD: wavelog # <- Insert a strong password here
    volumes:
      - wavelog-dbdata:/var/lib/mysql
    restart: unless-stopped

  wavelog-main:
    container_name: wavelog-main
    image: ghcr.io/wavelog/wavelog:latest
    depends_on:
      - wavelog-db
    environment:
      CI_ENV: docker
    volumes:
      - wavelog-config:/var/www/html/application/config/docker
      - wavelog-uploads:/var/www/html/uploads
      - wavelog-userdata:/var/www/html/userdata
    ports:
      - "8086:80"
    restart: unless-stopped

volumes:
  wavelog-dbdata:
  wavelog-uploads:
  wavelog-userdata:
  wavelog-config:
``

HB9HIL, 2024-06-11:
Hello @izeroo 

see here

[image]

izeroo, 2024-06-11:
@HB9HIL Thanks. I have not noticed that, my bad.

closed 2024-06-11 by izeroo