wavelog/wavelog #469: Gridsquare Map mixing SAT and others QSOs

ok1cdj, 2024-06-13:
How to filter out SAT QSOs on Gridsquare Map and show only terrestrial ? I think this worked before update.
[image]

int2001, 2024-06-24:
@ok1cdj since there's a new filter with "Propagation == empty" now, can we close this topic?

ok1cdj, 2024-06-24:
Hello, this is not working for me. Because i  am using TROPO or ES or MS in Propagation field.
I just need filter out all QSOs with propagation SAT.

Thank you very much. 73 Ondra

int2001, 2024-06-26:
Roger.
Added a new Select with #496 - could you pse test?

[image]

ok1cdj, 2024-06-26:
Great. Thanks.

closed 2024-06-26 by ok1cdj