wavelog/wavelog #475: V 1.6.1 - eQSL Download hängt

ds2k5, 2024-06-16:
Hallo,
habe die Version 1.6.1 im Einsatz.
Hab da ein Problem das ein eSQL von einem QSL hängt.
Ursache ist vermutlich das ich auf eQSL die QSL Karte nicht mehr sehe.
Wavelog diese aber gesehen hat und diese runterladen will.

Third party services -> eSQl ... -> Download eQSL Cards

[image]

[image]

Neuere eSQL Karten werden so auch nicht geladen.

Wie kann ich das beheben ?

int2001, 2024-06-19:
Du kannst sie manuell (Edit QSO) versuchen auf "No" zu setzen.
Dann sollte(!) sie auch nicht mehr versucht werden runterzuladen.

[image]

ds2k5, 2024-06-19:
@int2001

vielen Dank.... das ging soweit das ich
keine Fehler mehr hab

Es kamen aber keine eQSL Karten mehr rein..
aber gut versuch ich von hand runterladen und dann hochladen

ds2k5, 2024-06-20:
@int2001 

Nun habe ich ein eSQL Karte im Eingang bei eqsl.cc

[image]

aber Wavelog behauptet immer noch es gäbe keine eSQL Karten zum download

[image]

int2001, 2024-06-20:
Auch nicht wenn du erst auf "Download QSOs" drückst?!

ds2k5, 2024-06-20:
@int2001 
leider nein

int2001, 2024-06-20:
Und das entsprechende QSO existiert auch bei Dir im Wavelog und ist entsprechend zu eQSL hochgeladen worden?

Mit gleichen Angaben zu Zeit, Band und Mode?

Ansonsten schick mir bitte mal eine Mail, dann schau ich mir das im 1:1 mit Dir mal an, wenn Du magst und wir einen Termin finden.

Mailadresse unter qrz.com ersichtlich.

DJ7NT

ds2k5, 2024-06-20:
ja

[image]

ds2k5, 2024-06-24:
@int2001 vielen Dank für die Lösing:

[image]

Hier auf Import klcken

dann sind die QSL Karte zum DL da...

[image]

closed 2024-06-24 by ds2k5