wavelog/wavelog #516: QSO: Hardening callsign and locator input against unnecessary spaces at the beginning or end of the callsign

HochdruckHummer, 2024-06-29:
In Cloudlog and Wavelog, from time to time I'm suffering from QSOs not being saved, because I have entered unnecessary spaces at the beginning or end of the callsign or gridlocator.

This happens frequently when copying for example from QRZ and pasting into the text fields.

If then you try to save the QSO, nothing happens, not even an error message. I had been using Cloudlog for half a year before I found out the spaces are the problem. Before I found out i thought both systems were just buggy or a session had run out of time.

I can imagine other users are also suffering from that problem.

HB9HIL, 2024-07-05:
@HochdruckHummer 

Fixed in `dev` branch. Will be in the next release. Please feedback and close this issue if it's solved for you

HB9HIL, 2024-07-21:
@HochdruckHummer 

Hello
were you able to test? Can we close this issue?

HochdruckHummer, 2024-07-29:
I couldn't test in in time for 2 reasons:

- vacation
- never made an update on my instance before. Can you give me an advice on how to make the update? I have Wavelog running on my plesk webhosting server. Simply overwriting all files in the directory?

HochdruckHummer, 2024-07-29:
[image]

HB9HIL, 2024-07-29:
> Sorry, I couldn't test in in time for 2 reasons:
> 
> * vacation
> * never made an update on my instance before. But found the information on your page and updated.
> 
> I now installed v 1.7. issue still exists. Leading spaces still render gridlocator or callsign input wrong.
> 
> I'll try to go into detail, maybe it's not a space, but some other character, that breaks the input. Here is the user story to that issue.
> 
> Let's say I have a QSO with someone and I enter their callsign on QRZ.com. Then from their QRZ page, let's say https://www.qrz.com/db/do8ydp I copy the callsign by marking it back to front and then Ctrl+C it and enter it in Wavelog with Ctrl+V.
> 
> <img alt="Bildschirmfoto 2024-07-29 um 13 15 52" width="428" src="https://private-user-images.githubusercontent.com/60255369/353024791-ad329754-4a49-44c0-b07d-4e96ebf9919a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIyNTU1MDUsIm5iZiI6MTcyMjI1NTIwNSwicGF0aCI6Ii82MDI1NTM2OS8zNTMwMjQ3OTEtYWQzMjk3NTQtNGE0OS00NGMwLWIwN2QtNGU5NmViZjk5MTlhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzI5VDEyMTMyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVjYzc1MDQ1Yzk2ZDdiZWNjM2VmZjFkNTUwOGMyMmQzMTQ1NGYxZmE4MzhmYjI1ZGQ0ZmY0ZDgwYmY3YTU5MjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.vmjn1vJDbx_XZeDvWM91WvHz1uBAREjeqFpv02ZFrFc">
> It has a leading character.
> 
> Same with the grid locator from Detail page.
> 
> <img alt="Bildschirmfoto 2024-07-29 um 13 16 30" width="382" src="https://private-user-images.githubusercontent.com/60255369/353024898-504bbeba-f007-405f-b9fc-ce0f86e5b5f5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIyNTU1MDUsIm5iZiI6MTcyMjI1NTIwNSwicGF0aCI6Ii82MDI1NTM2OS8zNTMwMjQ4OTgtNTA0YmJlYmEtZjAwNy00MDVmLWI5ZmMtY2UwZjg2ZTViNWY1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzI5VDEyMTMyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVhZjQ3MmMwYjNhZWZjZjE4ZTRhY2M3MzM5OTM0MTE1MzhkOWI5OTMyMzZjNWZhNGIyNTU4Mzk0ZWM3YTQwOWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ZoSYMOkR6zIFXS4DvjW_4GtbwdV2J_775GgSUe1f3XM">

Where exactly do you paste it in?

HochdruckHummer, 2024-07-29:
In Post QSO input mask.

HB9HIL, 2024-07-29:
I also added a small commit to the `dev` branch for the search input mask [here](https://github.com/wavelog/wavelog/commit/218ea40b33ce329e5f7306eac1e8dbe0f6334fb2).

But anyway I tested your callsign and here it works. 

Reason why it probably does not work on your side is this method calls `trim()` (removing spaces at the beginning and end of a string) is happening in Javascript in this case. 

Your browser may cache the javascript and uses old code. 

Try to clear your browser cache/cookies or force him to reload ressources:

On Windows you press "Shift + F5"
On Mac you press "Cmd + Shift + R"

Please feedback after that

HochdruckHummer, 2024-07-29:
OK. I'm sorry. That was so dumb. Sorry for wasting your time. After deleting cahced ressoures, input automatically cuts of nonsense spaces.

BTW: I'm fully aware this was more a design issue on QRZ and other pages, instead of Wavelog. But I could imagine many people might have suffered from this, as I couldn't explain the behaviour myself, until I found the reason.

HochdruckHummer, 2024-07-29:
Seems fixed in 1.7.

closed 2024-07-29 by HochdruckHummer

HB9HIL, 2024-07-29:
> OK. I'm sorry. That was so dumb. Sorry for wasting your time. After deleting cached ressoures, input automatically cuts of nonsense spaces.
> 
> BTW: I'm fully aware this was more a design issue on QRZ and other pages, instead of Wavelog. But I could imagine many people might have suffered from this, as I couldn't explain the behaviour myself, until I found the reason.

No reason to say sorry :D And it was no time waste as I fixed it for the search bar aswell. Thanks for reporting this 👍 Such reports are really important for a good and stable user experience!

Thanks for using Wavelog :)

PR #517 (2024-07-01): [QSO] sanitize some input data
While waiting for the ferry after the HAMradio expo I fixed #516 

Please test and break