wavelog/wavelog #530: Feature Request: Antenna's for Station Locations

HB9HIL, 2024-07-01:
often heard and often requested: Ability to set an antenna/antennas for station locations. 

> And maybe adding the "default ANTENNA" as well?

_Originally posted by @wafer-li in https://github.com/wavelog/wavelog/issues/418#issuecomment-2144428575_

int2001, 2024-10-04:
Several Questions about `MY_RIG` are coming.
If you like @HB9HIL add it as well at the Loc

dm1ar, 2026-09-06:
Why not use the already defined namespace from SigMF for antenna data:

The following names are specified in the antenna namespace and should be used in the global object:

| Field | Required | Type | Short Description |
|---|---|---|---|
| `antenna:model` | Required | `string` | Antenna make and model number |
| `antenna:type` |  | `string` | Antenna type |
| `antenna:low_frequency` |  | `number` | Low frequency of operational range, in Hz |
| `antenna:high_frequency` |  | `number` | High frequency of operational range, in Hz |
| `antenna:gain` |  | `number` | Antenna gain in direction of maximum radiation or reception, in units of dBi |
| `antenna:horizontal_gain_pattern` |  | `array` | Antenna gain pattern in horizontal plane from 0 to 359 degrees in 1-degree steps, in units of dBi |
| `antenna:vertical_gain_pattern` |  | `array` | Antenna gain pattern in vertical plane from -90 to +90 degrees in 1-degree steps, in units of dBi |
| `antenna:horizontal_beam_width` |  | `number` | Horizontal 3-dB beamwidth, in degrees |
| `antenna:vertical_beam_width` |  | `number` | Vertical 3-dB beamwidth, in degrees |
| `antenna:cross_polar_discrimination` |  | `number` | Cross-polarization discrimination |
| `antenna:voltage_standing_wave_ratio` |  | `number` | Voltage standing wave ratio, in units of volts |
| `antenna:cable_loss` |  | `number` | Cable loss for cable connecting antenna and preselector, in dB |
| `antenna:steerable` |  | `boolean` | Defines if the antenna is steerable or not |
| `antenna:mobile` |  | `boolean` | Defines if the antenna is mobile or not |
| `antenna:hagl` |  | `number` | Antenna phase center height above ground level, in meters |