wavelog/wavelog #535: WavelogBashCat Broken Link

n1zzo, 2024-07-02:
The link to WavelogBashCat in the Wiki is broken, and the software is nowhere to be found in the web.
Is the tool deprecated? Can you point to any alternative to interface wavelog with rigctld?

Thanks,

Niccolò, IU2KIN

HB9HIL, 2024-07-02:
@n1zzo 

You may looking for https://github.com/g0wfv/CloudlogBashCat

There is no "WavelogBashCat"

BUT

We have WavelogGate which handles the Connection between FLrig and Wavelog

See here: https://github.com/wavelog/WaveLogGate

n1zzo, 2024-07-02:
Great,thank you so much for the quick reply, I’ll try to setup one of the two.Feel free to close the issue, thanks!Niccolò

HB9HIL, 2024-07-02:
Note: Adjusted Wiki Entry for "Radio-Interface" and fixed broken links.

closed 2024-07-02 by HB9HIL