wavelog/wavelog #542: Date format not the same in all views.

Nailcoding, 2024-07-05:
When setting the preferred date format in the account settings, this format is used in the logbook overview, but not when entering new QSOs (e.g. in Live-QSO).
The preferred date format should be used everywhere.

Wavelog V1.6.1

AndreasK79, 2024-08-25:
I agree that the date format should be used everywhere. It's not so easy to do in the QSO entry page.
An attempt was made in Cloudlog a few years back, but it turned out that the date was not saved correctly in the database with certain formats.

Perhaps @T012m3n7oR would be up for giving it a go in Wavelog?

sparkes, 2024-12-08:
A suggestion for a potential fix that would be a breaking change.  Note: I haven't looked at the source code at all so this might not be a solution at all.

Always store the date in the db in a std format and transform that on the client side or when communicating with third party api's to the preferred format.

M7 SPK - just taking a look at the software and saw the open issues tagged help wanted.

AndreasK79, 2025-09-01:
This is now implemented for QSO Entry in the current release.
Contest logging will have it in the next release.

closed 2025-09-09 by github-actions[bot]