wavelog/wavelog #549: LoTW upload failed: Signing failed

violarulan, 2024-07-09:
**Describe the bug**

My LoTW account won't update any new QSOs.

Cron works normally. If I click 'Manual Sync', it shows 'Signing failed.'.

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Rocky Linux 9.2 Apache/2.4.57 (Rocky Linux) OpenSSL/3.0.7
- PHP:             8.2.13 
- MySQL/MariaDB:   10.5.22-MariaDB
- Wavelog:          1.6
- tQSL        2.7.3

**Screenshots**
Include screenshots if applicable to illustrate the issue.

**Additional Context**

I've uploaded tQSL to latest version (2.7.3), re-exported my callsign certificate and uploaded it to Wavelog. It still shows the same.

After some searching, the message is at 

https://github.com/wavelog/wavelog/blob/d846bc153b90245010700a4ba0c426f2fdc8214f/application/controllers/Lotw.php#L275

If I'm right, Wavelog searches the ADIF file, if it contains `<SIGN_LOTW_V2.0:1:6>`, it stops uploading.

I print the `$adif_to_save` variable in the context and found every QSOs in the file contains `<SIGN_LOTW_V2.0:1:6>`.

```
// $adif_to_save variable
<TQSL_IDENT:54>TQSL V2.5.4 Lib: V2.5 Config: V11.12 AllowDupes: false

<Rec_Type:5>tCERT
<CERT_UID:1>1
<CERTIFICATE:1654>MIIEwDCCA6igAwIBAgIDCse1...............QEBCwUAMIHYMQswCQYDVQQGEwJV
.....
<EOR>

<Rec_Type:8>tSTATION
<STATION_UID:1>1
<CERT_UID:1>1
<CALL:6>BG2ELG
<DXCC:3>318
<GRIDSQUARE:6>PN17wd
<ITUZ:2>33
<CQZ:2>24
<IOTA:0>

<EOR>

<Rec_Type:8>tCONTACT
<STATION_UID:1>1
<CALL:6>YE1...
<BAND:3>20M
<MODE:3>FT8
<FREQ:9>14.076256
<FREQ_RX:1>0

<QSO_DATE:10>2024-07-09
<QSO_TIME:9>09:44:30Z
<SIGN_LOTW_V2.0:1:6>
<SIGNDATA:51>24PN17WD.....YE1...14.0762560FT82024-07-0909:44:30Z
<EOR>

<Rec_Type:8>tCONTACT
<STATION_UID:1>1
<CALL:6>VK7...
<BAND:3>20M
<MODE:3>FT8
<FREQ:9>14.076256
<FREQ_RX:1>0

<QSO_DATE:10>2024-07-09
<QSO_TIME:9>09:43:30Z
<SIGN_LOTW_V2.0:1:6>
<SIGNDATA:51>24PN17WD.....VK7...14.0762560FT82024-07-0909:43:30Z
<EOR>

<Rec_Type:8>tCONTACT
<STATION_UID:1>1
<CALL:6>BG4...
<BAND:3>20M
<MODE:3>FT8
<FREQ:9>14.076197
<FREQ_RX:1>0

<QSO_DATE:10>2024-07-09
<QSO_TIME:9>09:41:20Z
<SIGN_LOTW_V2.0:1:6>
<SIGNDATA:51>24PN17WD.....BG4...14.0761970FT82024-07-0909:41:20Z
<EOR>
...
```

I don't know how it works - during the debugging progress, I enabled Debug Mode (created a `.debug` file in the root folder). It uploads successfully but LoTW rejects it: 

```
// LoTW -> Activity
2024-07-09 10:07:06 LOTW_QSO: Processing file: 20240709100638.14106
2024-07-09 10:07:06 LOTW_QSO: User file: bg2elg-2024-07-09-10-06-37-wavelog.tq8
2024-07-09 10:07:06 LOTW_QSO: Certificate found for BG2ELG - CHINA (318)
2024-07-09 10:07:06 LOTW_QSO: Error in record 3: SIGN_LOTW_V2.0: ADIF field size overflow
2024-07-09 10:07:06 LOTW_QSO: Record 3 ignored due to errors
```

phl0, 2024-07-09:
Did you export the certificate without password?

int2001, 2024-07-10:
Without digging in deeper, did you take care of this one here?: https://github.com/wavelog/wavelog/issues/377

violarulan, 2024-07-10:
@int2001 
Yes. It was not the import issue that you mentioned. 
In this case, wavelog can handle the given .P12 cert. It shows up after uploading.

phl0, 2024-07-10:
Just for documentation purposes I just did a test with @int2001. I signed and uploaded to LoTW with SHA256 signature but this fails processing:

[image]

phl0, 2024-07-10:
I merged the PR so that at least the logs show that there was an error. I am in touch with LoTW to see/check if they would allow more recent hash algos.

phl0, 2024-07-29:
@violarulan Can we close here for now? The signature algo stuff is something that we cannot handle right now without LoTW accepting more recent algos ...

violarulan, 2024-07-29:
@phl0 Sure thing. The problem is solved. 

LoTW is too big and old to make such changes imo.

closed 2024-07-29 by violarulan

anthonydiiorio, 2024-07-29:
Still having this issue in the Docker version, I tried version 1.7 and dev.

Just tested now on dev with this result:
[image]

anthonydiiorio, 2024-08-02:
Upgraded to Docker version 1.8 and the issue remains for me

I tried updating tqsl and reuploading my p12 file but it still doesn't work

[image]
[image]
[image]

anthonydiiorio, 2024-08-03:
Verified on a clean Docker install tonight, is there a way we can enable legacy crypto in the Dockerfile?

[image]

HB9HIL, 2024-08-03:
Hello Anthony

The verification on the docker install does definitly work. 

Did you export the p12 certificate from TQSL WITHOUT a password?

HB9HIL, 2024-08-03:
I would say there is something wrong with your certificate..

anthonydiiorio, 2024-08-03:
I just tried the following:
- Reinstalled TQSL
- Imported my certificate into TQSL
- Export .p12 certificate from TQSL with no password
- Import .p12 into Wavelog

`Error in record 3: Unable to verify tCONTACT signature: QSOVerify failed - Verification failed`

It's strange because it was working for months before LoTW went down, and now I haven't been able to upload a log from Wavelog since it came back online. Perhaps something changed on their backend. If I export an adif and do "Sign a log and upload it automatically to LoTW" in TQSL it uploads just fine. Running the latest 2.7.3 TQSL.

If I go back through my LoTW activity the last successful one was:
`2024-05-12 02:01:01 | Web | ve2hew-2024-05-12-02-00-02-wavelog.tq8`

I am going to request a new certificate from the ARRL and see what happens.

HB9HIL, 2024-08-03:
Let us know after you uploaded the newly requested cert

anthonydiiorio, 2024-08-03:
It turns out I was chasing my tail with this, it seems to be related to my station location province. I opened a new issue here #689