wavelog/wavelog #562: LoTW upload fails for QSO with PROP_MODE unsupported by LoTW

dl3jop, 2024-07-11:
If you selected a PROP_MODE which is not supported by LoTW, the processing of the file fails on the LoTW server side.
List: [https://groups.arrl.org/g/ARRL-LoTW/topic/modes_that_can_be_submitted/26868793](url)

Steps to reproduce:
Create a QSO with propagaion set to RPT. The upload works, but LoTW will not process it.
The log on the LoTW Activity page is as follows:

LOTW_QSO: Processing terminated with an unexpected failure. System administrator has been notified.

It looks like all QSO in that upload are not processed, even if just one is set to one of the non supported propagation modes

dl3jop, 2024-07-11:
I think @phl0 is already looking for a fix :)

phl0, 2024-07-11:
Fix proposed in https://github.com/wavelog/wavelog/pull/563.

Loki1737, 2024-07-25:
Thank you @phl0 , that you are fixing this issue. I did run into the same issue, and did not realise it, until shortly.

closed 2024-07-29 by phl0