wavelog/wavelog #567: add "Mark printed" in QSL-Label-printing-dialog

dg9vh, 2024-07-11:
**What Feature is Missing?**
To shorten the workflow of printing QSL-Labels a button "mark printed" within the QSL-Label-printing-dialog, where you finaly start the printing of the labels before you chose the printer to use could potentially shorten the workflow when printing labels.

Actually you have to print the labels and then in a second step you have to switch over to the print queue and click there on "mark all as printed" (or similar). This sometimes has potential to be forgotten so that you sometimes have labels left in the queue that are already printed.

**Why is This Feature Important?**
I think the two factors above are the important things:
- saving time by having less mouseclicks
- saving from forgetting to mark them as printed.

**Is This Feature Personal or Beneficial to Others as Well?**
This feature would be a benefit for all those who are using the label printing functionality :-)

int2001, 2024-07-11:
Good idea. I am going to add a checkbox "mark as printed" to the printing dialog.

dg9vh, 2024-07-11:
I think internally you could use already exististing functions for this so that only some UI and command-routing has to be implemented :-)

int2001, 2024-07-21:
@dg9vh can we close this issue?

dg9vh, 2024-07-21:
Woeks as designed

closed 2024-07-21 by dg9vh