wavelog/wavelog #569: Docker new Language not working

zaubererty, 2024-07-11:
**Describe the bug**
Upgraded from a working 1.6 docker setup to 1.7 and got following error

[image]

**Version Information**
Docker container 1.7

HB9HIL, 2024-07-11:
Can you please try to restart the container again and delete browser cache?

meanwhile I will test an upgrade from 1.6 to 1.7

zaubererty, 2024-07-11:
Did recreate the whole stack, did not change anything

HB9HIL, 2024-07-11:
Looks your `docker-compose.yml` **exactly** like this?  :
https://github.com/wavelog/wavelog/wiki/Installation-via-Docker#docker-composeyaml

Did you upgrade like this:

Stop the container:
```bash
docker compose down
```

Repull
```bash
docker compose pull
```

Restart the container
```bash
docker compose up -d
```

HB9HIL, 2024-07-11:
I just tested an upgrade from 1.6 to 1.7
works as expected here

zaubererty, 2024-07-11:
My docker-compose.yml looks like this

`version: '3'

services:
  web:
    image: ghcr.io/wavelog/wavelog:latest
    restart: unless-stopped
    networks:
      - external_network
      - internal_network
    depends_on:
      - mariadb
    volumes:
      - ${PATH_TO_APPDATA}/wavelog/storage/config:/var/www/html/application/config/docker
      - ${PATH_TO_APPDATA}/wavelog/storage/logs:/var/www/html/application/logs
      - ${PATH_TO_APPDATA}/wavelog/storage/backup:/var/www/html/backup
      - ${PATH_TO_APPDATA}/wavelog/storage/uploads:/var/www/html/uploads
      - ${PATH_TO_APPDATA}/wavelog/storage/userdata:/var/www/html/userdata
    ports:
      - 8082:80
  mariadb:
    restart: unless-stopped
    image: mariadb:11.3
    networks:
      - internal_network
    volumes:
      - ${PATH_TO_APPDATA}/wavelog/db:/var/lib/mysql
  adminer_container:
    image: adminer:latest
    ports:
      - 8089:8080
    networks:
      - internal_network
      - external_network

networks:
  external_network:
  internal_network:
    internal: true
`
What files need to be in the docker specific config folder?

HB9HIL, 2024-07-11:
@zaubererty This is far away from the default setup. I'm no docker expert. Maybe @int2001 can help here.

But I'm 99% sure that this is no Wavelog Code Issue. 

> What files need to be in the docker specific config folder?

only `config.php` and `database.php`

ALL other files are laying in `application/config`

One way to solve that is delete all config files in `application/config/docker` except the two mentioned

zaubererty, 2024-07-11:
Got it working, thank you for the tips

closed 2024-07-11 by zaubererty