wavelog/wavelog #584: Contest Logging - radio won't be saved as default.

205er, 2024-07-15:
**Describe the bug**
The radio (hardware) linkage won't be saved for (Live) Contest logging. As soon as you leave the logging page and access it again the previously selected radio is no longer chosen

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- PHP:              8.2.21
- Wavelog      1.7

**Expected Behavior**
I would expect the radio selection to be restored after accessing the contest log again.

Thanks!

205er, 2024-07-17:
Unfortunately I discovered today, that the station setting gets lost even for the "Live QSO"-logging. In this case only after one has logged on again. This means leaving the logging page and coming back later is no problem (station setting is kept), but it gets reset after logon.

HB9HIL, 2024-07-17:
> Unfortunately I discovered today, that the station setting gets lost even for the "Live QSO"-logging. In this case only after one has logged on again. This means leaving the logging page and coming back later is no problem (station setting is kept), but it gets reset after logon.

See #418

AndreasK79, 2024-07-18:
I noticed that contest logging does not set radio in user session either. It does that in regular QSO entry. I added an update for this when logging QSOs in the contest part.

Contest logging does set radio, but only if user session has it set, and that happens when you log in the regular QSO entry.,

See #592.

closed 2024-08-07 by HB9HIL