wavelog/wavelog #601: Advanced Logbook Map Showing Incorrect Distance Unit

thecocohead, 2024-07-21:
**Describe the bug**
On the Advanced logbook page, the Distance shown on the map when hovering over is always in Kilometers, regardless of the user's preferred measurement unit. 

**Version Information**
- OS: Ubuntu 22.04
- PHP: v8.1.2
- MySQL/MariaDB: v8.0.37
- Wavelog: 1.7 (commit 8a9a0f09)

**Steps to Reproduce**
1. Go to Logbook > Advanced
2. Click on the "Map" Button
3. Observe "Distance" on the bottom of the screen

**Expected Behavior**
The distance shown should be in the user's measurement unit. 

**Screenshots**
[image]

**Desktop Information:**
- OS: Win10 22H2
- Browser: Firefox
- Version: 128

**Additional Context**
I'll be working to fix this issue and submitting a PR when I find a fix.

thecocohead, 2024-07-21:
I think I've narrowed down the source of this bug. 
In assets/js/leaflet/geocoding.js line 61-74, the function `onMapMove` does not use the user's preference in distance when calculating and displaying distance. Here's the code snippet:
```js
function onMapMove(event) {
	var LatLng = event.latlng;
	var lat = LatLng.lat;
	var lng = LatLng.lng;
	var LatLng2 = ConvertDDToDMS(lat, lng);
	$('#latDeg').html(LatLng2.latDeg);
	$('#lngDeg').html(LatLng2.lngDeg);
	var locator = latLngToLocator(lat,lng);
	$('#locator').html(locator);
	var distance = bearingDistance(homegrid, locator);

	$('#bearing').html(distance.deg + ' deg');
	$('#distance').html(Math.round(distance.km * 10) / 10 + ' km');
};
```
This code is used in a few other locations inside Wavelog:
- Analytics > Gridsquare Map
- Analytics > Activated Gridsquares
- Various Gridmaster awards

All are affected. 
***
As for a fix, I'm not sure what that looks like. 
- We can make ajax calls to get the user's preference, but making 30+ calls per second doesn't sound like it'd be worth it performance wise. 
- We could store the result of a single ajax query into a global variable, but the new global variable won't change if the user changes their distance preference and doesn't close the tab. 

Gonna leave this to over contributors/maintainers to see what the best course forward is- even if that course is "won't fix".

AndreasK79, 2024-07-21:
Hi. Thanks for reporting. When I added this a while back, I never thought about the user settings for distance. We will certainly see if we can fix this issue.

AndreasK79, 2024-07-21:
I have a working solution now. Need to test it a bit more. Will let you know when it's ready.

AndreasK79, 2024-07-22:
#605 added

AndreasK79, 2024-07-24:
This is merged into dev, so I am closing this one now.

closed 2024-07-24 by AndreasK79