wavelog/wavelog #620: DX Colander HTTP Error 500

jwussler, 2024-07-24:
The DX Calendar under the menu twisty is producing a 500 error

i can goto the wavelog.org demo and get the same error 

PHP Version | 8.3.9 OK
MySQL Version | 11.3.2-MariaDB-1:11.3.2+maria~ubu2204
Codeigniter Version | 3.2.0-dev

int2001, 2024-07-24:
Seems like the DXPedition to Mariana Island causes this trouble.
The DXPedition is across new years eve. So it starts in 2024 and ends in 2025.
Since the Date isn't really well formatted it can currently not be parsed well by Wavelog.

We're trying to fix that. CC: @AndreasK79

int2001, 2024-07-24:
pse have a look at the current `dev`. should be fixed.
Would be great if you can confirm and close this issue @jwussler

jwussler, 2024-07-24:
Everything works now,

Thanks!