wavelog/wavelog #627: QSO are marked as sent to LotW, even if the LOTW Import was not accepted

Loki1737, 2024-07-25:
**Describe the bug**
I am using Wavelog to automatically transfer the Logfiles to LotW. In some cases, these log files are not accepted. For example, if one of the QSO was made via an unsopported propagation method (check https://github.com/wavelog/wavelog/issues/562#event-13482481772). But nevertheless, all the QSO are marked in this scenario as "sent to LotW". And as they are sent, It will never be tried again to resent them. Therefore a lot of QSOs in my logbook ended up as not being sent to LotW, because one QSO does had the Propagation set to Repeater.

It is also quite similar for errors, where for example the correct certificates were not uploaded: In LotW the error ends with:
2024-05-11 18:14:02 LOTW_QSO: Error in record 3: Unable to verify tCONTACT signature: QSOVerify failed - Verification failed

But nevertheless the QSOs was marked as sent.

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Apache
- PHP:              8.3.8
- MySQL/MariaDB:   10.5.25-MariaDB-ubu2004-log
- Wavelog:          1.7

**Steps to Reproduce**
1. Create a QSO via Repeater (prior to version 1.7.1), or delete the LotW certificate
2. Create a second repeater, using the same callsign
3. Send them manually to LotW
4. Search LotW for these QSOs

**Expected Behavior**
I would expect, that an error is generated, and the QSO is not marked as sent to LotW

**Additional Context**
I made the error, that I uploaded the certificates after I set up (and tested) the LotW certificates. Therefore I thought, that I thought, that it was uploaded, but the QSOs were never actually uploaded to LotW. To fix it, I had to edit all QSOs manually.

phl0, 2024-07-25:
This has been dealt with in https://github.com/wavelog/wavelog/pull/573 which is in dev branch already. Which branch are you using? The INTERNET and RPT prop mode QSOs are now ignored in LoTW upload.

Loki1737, 2024-07-25:
@phl0 I am not using dev. Therefore I missed that part. I am on Version 1.7

But the propagatiom was only a part of it. Of course the one, which happens more frrwurntly if everything is set up pproperly. 

My issue was the observability at all. When I did not set up the certificates properly and LotW send an error, wavelog updates the qsos still as uploaded to LotW. 

And if I understood #573  correctly, it is also not updating Any set of qsos in the past, which were send with the same file as one unsupported aso, and therefore did not end up in the database from LotW. And they never will, as they are sent as marked. 

Would you prefer if I wait for the next version and check the behavior afterwards? And reopen the issue in the case, if I still miss some aspect?

phl0, 2024-07-25:
The problem is we do not know if/when a single QSO is rejected. So currently I have no idea how to catch that....

Loki1737, 2024-07-25:
Okay, I got that. This means it is at first a feature request to the ARRL, to provide that option.

Would it be possible, to add at least the LotW Send Status to the Batch edit in the Advanced Logbook? This would at least allow me, to batch edit, in case, if I am running into issues.

phl0, 2024-07-25:
That should be feasible. As for LoTW we only know if the file itself got uploaded. The file is processed at a later stage and separately. So we do not get any feedback from the processing.

@AndreasK79 U in for the LBA change?

Loki1737, 2024-07-25:
I added a section in the wiki, to document the known issue:
https://github.com/wavelog/wavelog/wiki/LoTW-Import-&-Export-Documentation

Loki1737, 2024-07-25:
And I just created a ticket on LotW, with the feature request. I will let you know, if I get any feedback.

phl0, 2024-07-25:
> And I just created a ticket on LotW, with the feature request. I will let you know, if I get any feedback.

You have a link to the ticket or the like?

Loki1737, 2024-07-25:
> > And I just created a ticket on LotW, with the feature request. I will let you know, if I get any feedback.
> 
> You have a link to the ticket or the like?

Unfortunatly only an E-Mail.

AndreasK79, 2024-07-25:
> @AndreasK79 U in for the LBA change?
Sure thing.

AndreasK79, 2024-07-26:
PR #636 for LoTW sent/received in batch edit.

AndreasK79, 2024-07-26:
@Loki1737 batch edit for LoTW is merged into dev now.