wavelog/wavelog #629: qrz.com password does not get encoded during installation

nobswolf, 2024-07-25:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**

Password of qrz.com accepts chars that result in a broken installation

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu

- OS:               Debian 12
- Wavelog:          1.7 (current git) 
- Server Software 	Apache
- PHP Version 	8.2.20 OK
- MySQL Version 	10.11.6-MariaDB-0+deb12u1
- Codeigniter Version 	3.2.0-dev

Hint: it would be nice to have this as a ready-to-copy block in the debug menu

**Steps to Reproduce**

1. install wavelog and start installer
2. enter qrz.com credentials containing a '  (single quote) 
3. wavelog hangs after the installer really starts
4. find error in Apache error.log like:

... AH01071: Got error 'PHP message: PHP Parse error:  syntax error, unexpected identifier "D" in /var/www/html/wavelog/application/config/config.php on line 46; PHP message: PHP Parse error:  syntax error, unexpected identifier "D" in /var/www/html/wavelog/application/config/config.php on line 46'

**Expected Behavior**
Either do not accept passwords or better escape them accordingly

HB9HIL, 2024-07-26:
Hello @nobswolf 

Certain special characters are always an issue in password, not only in Wavelog. Therefore I added a js-check in the installer upgrade to prohibit `' / \ < >`

nobswolf, 2024-07-28:
The problem is: When you save a password of another service and you forbid chars that they allow, it will fail. The better way is to "escape" the difficult chars. There are several ways to do it, like:

https://www.php.net/manual/de/function.htmlspecialchars.php 

Another point is that you save the passwords in the PHP-source. Better to put them in the database. With just a little error in the configuration the PHP source is publicly readable. The database is a little protection against this.

HB9HIL, 2024-07-28:
Hello @nobswolf  

> The problem is: When you save a password of another service and you forbid chars that they allow, it will fail. The better way is to "escape" the difficult chars. There are several ways to do it, [...]

We know how to escape things in PHP :D Reason I decided to do the JS check yet is that the code in Wavelog needs some rework regarding hamqth and qrz passwords (and other 3rd party services). Plan is to move that to the DB but until then we have to live with the JS check for the moment.

HB9HIL, 2024-07-28:
@nobswolf 

Let's create a new issue for moving passwords from config.php to database and close this one as installer is "fixed" currently. Agree?

PR #603 (2024-07-28): [INSTALLER] Next generation of the Wavelog Installer
First I just wanted to fix some known bugs in the installer after the big Frontend Remake in #37 

But I wanted to improve the backend aswell and give the user a better feedback during the process as the whole database stuff can take a while. So I rebuilt the backend of the installer and added some frontend stuff for better response to the user. 

Improvements:

- Fixed some bugs regarding language
- Save inputs in localStorage to don't loose all the entered data in case of a page reload
- Finally added the checklist on the last page
- Added [... cut]
closed 2024-08-05 by HB9HIL