wavelog/wavelog #631: max_upload_file_size should read

nobswolf, 2024-07-25:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
the php.ini attribute for the file size for uploads is spelled wrong

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:                Debian 12
- PHP:              8.2
- MySQL/MariaDB:     10.11
- Wavelog:          1.7 git

**Steps to Reproduce**
1. start installer
2. read about max_upload_file_size 
3. don't find it in php.ini
4. find  upload_max_filesize instead

**Expected Behavior**
correct name in the check

phl0, 2024-07-25:
Fix proposed in https://github.com/wavelog/wavelog/pull/632.

phl0, 2024-07-25:
Correction: Will be fixed with https://github.com/wavelog/wavelog/pull/603.

HB9HIL, 2024-07-28:
@nobswolf 

is fixed in debug view and installer. please check and close this issue :)

closed 2024-08-07 by HB9HIL