wavelog/wavelog #633: Make the list of DXCC countries full text searchable

nobswolf, 2024-07-25:
**By submitting feature requests, you have the opportunity to contribute your ideas and inspiration to the Wavelog Dev-Team. Please be aware that not every feature request will be addressed or answered.**

**What Feature is Missing?**
The list of the DXCC countries during installation can only get searched by the beginning of the string. For example: "GER" only finds the two deleted entries but not the correct one starting with "Federal".

**Why is This Feature Important?**
It should be easy and comfortable to find countries by their commonly used name and not only by the official one.

**Is This Feature Personal or Beneficial to Others as Well?**
I guess most people address countries by their simple name and not the official one containing the form of state.

HB9HIL, 2024-08-05:
Todo:

- [ ] Add searchable DXCC List in Installer
- [ ] Add searchable DXCC List in Station Location
- [ ] Add searchable DXCC List in Station QSO Logging

HB9HIL, 2024-08-16:
Hello @nobswolf 

The searchable DXCC dropdown is now available in the most important places. 
Please close this issue if it matches your expectations :) 👍

HB9HIL, 2024-08-20:
@nobswolf 

The DXCC dropdown is full text searchable in the installer aswell. This was done here: https://github.com/wavelog/wavelog/pull/715

In fact it was searchable even before that because you can make the dropdown active and type "switzerl" and Switzerland was selected. Now you have also a GUI for that with a fancy searchbar

closed 2024-09-04 by HB9HIL