wavelog/wavelog #647: Contest Calendar showing dupes

jwussler, 2024-07-28:
Server Software | Apache/2.4.58 (Ubuntu)
PHP Version | 8.3.6 OK
MySQL Version | 10.11.8-MariaDB-0ubuntu0.24.04.1
Codeigniter Version | 3.2.0-dev

I am on the dev branch.

On the contest calendar under the menu twisty im getting 3 dupes on most items under the weekend view

I can also log into the demo site at wavelog.org and the same results are there.

Thanks,

Joe
WAØO
 
[image]

AndreasK79, 2024-07-28:
Hi Joe. Thanks for reporting the issue. I have fixed this in the latest dev branch. If you are using dev, just to a pull, if not wait for the next release.

Andreas.

jwussler, 2024-07-28:
Looks like it works here now!

Thanks!

closed 2024-07-28 by jwussler