wavelog/wavelog #650: Spotting to DX-Cluster from Wavelog

dg9vh, 2024-07-28:
**What Feature is Missing?**
It would be nice to have a functionality to spot a contact into the dx-cluster (maybe like the toot or tweet function).

**Why is This Feature Important?**
If you are working with wavelog you have to open a second browser-tab to open a dxcluster and make a manual spot there. This could be errorneous in point of frequency or callsign because of doing it manually and nobody's perfect :-)

**Is This Feature Personal or Beneficial to Others as Well?**
I think everyone could profit from this.

int2001, 2024-07-29:
Hi Kim,
nice idea. But there is an unsolved problem with that.

Most of the huge Clusters aren't using an authentication. So if you spot sth. via telnet (that's the protocol in the background) you simply state your call and spot someone. e.g.: I can easily spot sth. on behalf of DG9VH. It'll occure in the cluster. But i risk my own connection to the cluster.

The only thing which exists is: If there are too many abuses, the Cluster-owner blocks your (in this case the instances) IP-Adress. After that no one can spot.

dg9vh, 2024-07-29:
Closed as withdrawn.