wavelog/wavelog #655: Missing cron table after fresh installation

yuko1225, 2024-07-29:
**Describe the bug**
Certain page cannot be open, for example, Debug page.

I enable the error logging, and this was the error:
Severity: error --> Exception: Table 'wavelog.cron' doesn't exist /.../system/database/drivers/mysqli/mysqli_driver.php 301

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               CentOS
- PHP:              8.1.29
- MySQL/MariaDB:    MariaDB 10.4.34
- Wavelog:          master?

**Additional Context**
I was following the wiki installation guide, maybe I was using the wrong branch? I can't find the cron table definition inside install/assets/install.sql either.

yuko1225, 2024-07-29:
> Hello @yuko1225
> 
> this sounds like the database migrations or at least some of them failed. Please answer the following questions so we can figure out what happened here:
> 
> Did you install via `git` or manually with downloading the zip file?
> 
> If you installed via `git` go to the directory of your Wavelog Installation and perform a `git branch -vv` and copy the output in here.
> 
> Please check the value of `$config['migration_version']` in the file `application/config/migration.php`. Don't change any value here!
> 
> If you have mysql access, show us the output of the following command:
> 
> `SELECT * FROM migrations`
> 
> You installed Wavelog on `CentOS`. Did you had any warnings in the installer?

1. It is a fresh installation, I gave up on migration from cloudlog(it was too missing something after migration). I used the git method.  Here is the output of git branch -vv:

> * master 8a9a0f0 [origin/master] Merge pull request #553 from wavelog/dev

2. I tried to checkout to the v1.7 tags, it didn't do anything.
3. $config['migration_version'] = 208;
4. 
> MariaDB [wavelog]> SELECT * FROM migrations;
>
> +---------+
> | version |
> +---------+
> |     195 |
> +---------+
> 1 row in set (0.001 sec)

5. No warning during installation, although it didn't went through the first time because I had InnoDB engine disabled on MariaDB. I re-enable the engine and went through the install wizard without any warnings. It is worth to notice that install wizard didn't check if the InnoDB is present.

HB9HIL, 2024-07-29:
Please turn on Error Logging (log_threshold) to Level 2 (Debug) and then call the page [WAVELOG_URL]/index.pgp/migrate

Check what you see in the Browser and the Error Log in application/logs/....

yuko1225, 2024-07-29:
> Please turn on Error Logging (log_threshold) to Level 2 (Debug) in your `application/config/config.php` and then call the page `[WAVELOG_URL]/index.php/migrate`
> 
> Check what you see in the Browser and the Error Log in `application/logs/`

Here is what I got:

> DEBUG - 2024-07-29 21:18:45 --> Migration process is currently locked. Second migration attempt ignored.
> DEBUG - 2024-07-29 21:18:50 --> Migration process is currently locked. Second migration attempt ignored.
> DEBUG - 2024-07-29 21:18:55 --> Migration process is currently locked. Second migration attempt ignored.
> DEBUG - 2024-07-29 21:18:56 --> No URI present. Default controller set.
> DEBUG - 2024-07-29 21:18:56 --> Migration process is currently locked. Second migration attempt ignored.
> DEBUG - 2024-07-29 21:18:56 --> Migration process is currently locked. Second migration attempt ignored.
> DEBUG - 2024-07-29 21:18:56 --> Migration class already loaded. Second attempt ignored.
> DEBUG - 2024-07-29 21:18:56 --> Migration process is currently locked. Second migration attempt ignored.
> DEBUG - 2024-07-29 21:19:00 --> Migration process is currently locked. Second migration attempt ignored.
> DEBUG - 2024-07-29 21:19:02 --> Migration process is currently locked. Second migration attempt ignored.
> DEBUG - 2024-07-29 21:19:02 --> Migration class already loaded. Second attempt ignored.
> DEBUG - 2024-07-29 21:19:02 --> Migration process is currently locked. Second migration attempt ignored.
> ERROR - 2024-07-29 21:19:02 --> Severity: Warning --> Undefined property: stdClass::$user_language /.../application/controllers/User.php 795
> DEBUG - 2024-07-29 21:19:02 --> Migration process is currently locked. Second migration attempt ignored.
> DEBUG - 2024-07-29 21:19:03 --> Migration process is currently locked. Second migration attempt ignored.
> DEBUG - 2024-07-29 21:19:03 --> Migration process is currently locked. Second migration attempt ignored.
> DEBUG - 2024-07-29 21:19:05 --> Migration process is currently locked. Second migration attempt ignored.
> DEBUG - 2024-07-29 21:19:05 --> Migration process is currently locked. Second migration attempt ignored.
> DEBUG - 2024-07-29 21:19:05 --> Migration class already loaded. Second attempt ignored.
> DEBUG - 2024-07-29 21:19:05 --> Migration process is currently locked. Second migration attempt ignored.
> DEBUG - 2024-07-29 21:19:10 --> Migration process is currently locked. Second migration attempt ignored.
>

HB9HIL, 2024-07-29:
please delete the file `application/cache/.migration_running` and try the same again.

After that check if you can call the Debug Page (Admin > Debug Information)

yuko1225, 2024-07-30:
> please delete the file `application/cache/.migration_running` and try the same again.
> 
> After that check if you can call the Debug Page (Admin > Debug Information)

Okay. So I found the cause after remove this lock file.

> DEBUG - 2024-07-30 02:08:13 --> Created migration lockfile
> DEBUG - 2024-07-30 02:08:13 --> Migrating up from version 195 to version 196
> ERROR - 2024-07-30 02:08:13 --> Severity: error --> Exception: Specified key was too long; max key length is 1000 bytes /home/cloudlog/system/database/drivers/mysqli/mysqli_driver.php 301

Apparently it was still related to the InnoDB, while I did enable the InnoDB I still kept the default engine as MyISAM. And it doesn't support key that over 1000 bytes long.

yuko1225, 2024-07-30:
> please delete the file `application/cache/.migration_running` and try the same again.
> 
> After that check if you can call the Debug Page (Admin > Debug Information)

> please delete the file `application/cache/.migration_running` and try the same again.
> 
> After that check if you can call the Debug Page (Admin > Debug Information)

Okay. So I found the cause after remove this lock file.

> DEBUG - 2024-07-30 02:08:13 --> Created migration lockfile
> DEBUG - 2024-07-30 02:08:13 --> Migrating up from version 195 to version 196
> ERROR - 2024-07-30 02:08:13 --> Severity: error --> Exception: Specified key was too long; max key length is 1000 bytes /.../system/database/drivers/mysqli/mysqli_driver.php 301

Apparently it was still related to the InnoDB, while I did enable the InnoDB I still kept the default engine as MyISAM. And it doesn't support key that over 1000 bytes long.

HB9HIL, 2024-07-30:
@yuko1225 

There are sooooo many possible configurations and setups out there. We have a recommendation for a working setup. Installing on CentOS with a custom MySQL configuration is beyond the possible check coverage. 

Did you manage to install Wavelog properly now by setting InnoDB as default?

yuko1225, 2024-07-30:
@HB9HIL Yes it looks like it is working fine now. I understand my setup is atypical, still it wouldn't hurt to note on installation guide that InnoDB is required on wavelog. I'm pretty sure it wasn't required on Cloudlog.

closed 2024-08-01 by yuko1225