wavelog/wavelog #656: Names vs. Keys on contest select lists

dg9vh, 2024-07-29:
**Describe the bug**
On different places different representations for the contests are in the select-lists:
On the contest log window you'll find the names as you'll find them on the contest-calendar (for most of them, we have to put some hands on this maybe to be synchronized with this):
[image]

If you switch over to Cabrillo export, you'll find the contests with their keys form the contests-table:
[image]

Now, if you go to the edit-window for QSO details, you'll find again the names:
[image]

This is also on batch edit in advanced log:
[image]

But here is the problem (did not test it on the edit above), that if you would change a contest to another name, the contest entry is completely deleted.

Running on Wavelog 1.7 Release version.

AndreasK79, 2024-07-30:
#663 added. Now uses name in Cabrillo export if available, if not id as before.

What do you mean when you say that the contest entry is completely deleted?

AndreasK79, 2024-07-31:
@dg9vh fix is in dev, so this should be ok now.

closed 2024-08-02 by AndreasK79