wavelog/wavelog #683: Keep me logged Checkbox not available in 1.8?

filipmelik, 2024-08-02:
Hi, 
first of all, thanks  for this great tool. 

Just one question - I have updated to 1.8 today, but I fail to see the "Keep me logged Checkbox" on the login screen. I was frustrated by the constant logouts, so I welcome this feature very well, but I cannot see it, even though the menu shows the version installed on my server is really an 1.8. 

I have also searched the 1.8 release's commits, but I could not see any relevant commit. Is it possible this feature somehow slipped through this release, or am I just blind? 😄  

Thanks and 73, Filip OK1GOD

HB9HIL, 2024-08-02:
Hello Filip

This feature requires two things:

- Check if you see any warnings in debug view. You may see a warning that your encryption key is still default.

- Your Wavelog Instance has to use HTTPS to activate the "Keep me logged in" feature.

filipmelik, 2024-08-03:
Thanks for quick reply!

I have my instance backed by HTTPS, so it was really issue with the default encryption key in the config file, so I changed it. 

However, the thing came to my mind when I was changing the config value - the description for the encryption key in the [config file](https://github.com/wavelog/wavelog/blob/master/application/config/config.sample.php#L397) is as follows:
```
If you use the Encryption class, you must set an encryption key. See the user guide for more info.
```

So when I was installing my instance and I was tuning the config, this description did not indicate that I really need to change it. It is probably some generic codeigniter framework's (which I personally do not know) description. So this could be improved by changing the config sample's description or alternatively documenting this "Keep me logged in" feature in the wiki 👍 

Thanks again!

HB9HIL, 2024-08-03:
@filipmelik You're right :) I'm gonna change the description :)

closed 2024-08-03 by HB9HIL